Bouni/python-luxtronik #6: Wrong data type Circulation_Pump

Volti-Ba, 2019-11-03:
241: Unknown("Circulation_Pump") is a Percent value (cerculation pump, no conversion needed) but has an division by 10
https://github.com/Bouni/luxtronik/blob/master/luxtronik/calculations.py#L244

Bouni, 2019-11-04:
Shit, I hoped that AIT is at least a little bit consistent in their use of units. So they have Percent values without divider and some with :man_facepalming: 
Will fix that soon ...

extin, 2019-12-03:
Also, atleast on MSW2-6S, pump that is shown on internal datapage is 173: Bool("ID_WEB_Durchfluss_WQ"), which has Bool as datatype instead of Flow.
https://github.com/Bouni/luxtronik/blob/5f7a206876020b361001da9678f89ff4922da23d/luxtronik/calculations.py#L177

Bouni, 2020-05-04:
@Volti-Ba Finally found some spare time to fix this. Maybe not the nicest way to introduce a Percent2 class but I think thats what it needs to overcome AITs pile of s***

closed 2020-05-04 by Bouni

Volti-Ba, 2020-05-04:
This need´s also Percent2  ;-.))

241: Percent("Circulation_Pump") is a Percent value (circulation pump, no conversion needed) but has an division by 10.
https://github.com/Bouni/luxtronik/blob/master/luxtronik/calculations.py#L244

commit 354f1f2e (2020-05-04): Fixed #6