Bouni/python-luxtronik #7: ID_WEB_Durchfluss_WQ is flow, not Bool

Bouni, 2019-12-03:
Also, atleast on MSW2-6S, pump that is shown on internal datapage is 173: Bool("ID_WEB_Durchfluss_WQ"), which has Bool as datatype instead of Flow.
https://github.com/Bouni/luxtronik/blob/5f7a206876020b361001da9678f89ff4922da23d/luxtronik/calculations.py#L177

_Originally posted by @extin in https://github.com/Bouni/luxtronik/issues/6#issuecomment-561241180_

remyderuysscher, 2020-02-02:
Correct this is a boolean. 

sensor.luxtronik_id_web_durchfluss_wq | True

macrojames, 2020-05-02:
I get 
```
INFO     - 173 : 592
```
which matches
[image]
on my alpha innotec on the websocket Interface

Bouni, 2020-05-04:
@macrojames Can you tell me what type of heatpump you have?

Bouni, 2020-05-04:
@remyderuysscher can you clarify if id_web_durchfluss_wq is actually a boolean or fow?

macrojames, 2020-05-04:
That's an alpha innotec lwcv 82r1/3
From the device:

Anlagenstatus
--
Wärmepumpen Typ | LW 8
Wärmepumpen Typ | CMD_6
Softwarestand | V3.84.3 (r5195)

Bouni, 2020-05-04:
The hard part begins here, as we no longer get names for the values out of the JAVA source (as does for all the others so far).
Any suggestions on how to name this value?
Maybe just flow_rate? or better 254_flow_rate in case we get another flow rate in the future?

remyderuysscher, 2020-05-04:
Hi bouni,

There is an issue with the IP address check. It crashes on start of HA,
I've posted a comment on the GitHub of the HA integration. Can you take a
look?

Regards Remy

Bouni, 2020-05-04:
@remyderuysscher Please lways open a new issue, avoid hijacking other issues, that way it is a nightmare to find things at a later point in time!

remyderuysscher, 2020-05-04:
Thanks will do! Sorry.

macrojames, 2020-05-04:
> The hard part begins here, as we no longer get names for the values out of the JAVA source (as does for all the others so far).
> Any suggestions on how to name this value?
> Maybe just flow_rate? or better 254_flow_rate in case we get another flow rate in the future?

I did some digging and found some suggestions in the firmware. I only looked for calculations yet and not all of them appear to have human readable names.
[image]
I drop this here. It contains some names or suggestions for them. I am not sure how to proceed here. I see that value 254 is not calculated differently than 173, just a little more logic around it. 173 would be called "Durchfluss", Type int.

WhistleMaster, 2020-05-05:
Same here on heat pump Alpha Innotec SWC 82K3. ID_WEB_Durchfluss_WQ should not be a Bool.

Bouni, 2020-05-06:
@macrojames 
What source did you look at to reverse the meaning of the variables?
I would highly appreciate if you can share it so that I can work on improving this module further!

closed 2020-05-06 by Bouni

macrojames, 2020-05-06:
I am not sure how far I am allowed to share my work.
I took the firmware from https://www.heatpump24.com/DownloadArea.php?layout=1 and extracted it until there was a file "appl".
That file could be interpreted by [Ghidra](https://github.com/NationalSecurityAgency/ghidra) really nice and the memory labels are human readable.
There is also a hardcoded root password in shadow but i guess you would need UART to access a console.

Bouni, 2020-05-06:
@macrojames 
Nice! I never used Ghidra before but its defenitely on my list 😄 
I already know about the root password and how to access the controller via SSH. If you want to know, send me an email to bouni@owee.de an I'll share my findings with you!

remyderuysscher, 2020-05-08:
> 
> 
> @remyderuysscher Please lways open a new issue, avoid hijacking other issues, that way it is a nightmare to find things at a later point in time!

I can confirm that the IP address issue is now solved! Thanks.

commit 44eefb84 (2020-05-06): Fix #7