Bouni/python-luxtronik #13: Wrong status of heatpump

remyderuysscher, 2020-05-10:
Hi @Bouni statuszeile1, 2 and 3 are wrong.  (ID_WEB_HauptMenuStatus_Zeile3)

It always shows that the heatpump is running (zeile1) and heating (zeile3) while the heatpump is idle or cooling as reported by the web interface (webbased version).

Bouni, 2020-05-10:
I kind of know about this issue and need to take a deeper look into it.
AIT completely messed this one up, the have a lot of nested if / else in their code to figure out what status to show.

Bouni, 2020-06-29:
I looked a bit closer into this and duped all data from my heatpump 3 times with different settings as displayed in the AlphaApp (Android App):

1. Heating off, hot water automatic, cooling off
2. Heating on, hot water automatic, cooling off
3. Heating off, hot water automatic, cooling on

This gives me the following conclusions:

- ID_WEB_HauptMenuStatus_Zeile{1-3} do not change at all when I change the settings

- Parameter 3 "ID_Ba_Hz_akt" shows the setting of the heating, it switches from Off to Automatic if I change it in the app
- Parameter 779 "ID_Ba_Hz_MK3_akt" does the same, at least for my heatpump model

- Parameter 4 "ID_Ba_Bw_akt" seems to show the setting of the hot water

- Parameter 108 "ID_Einst_BA_Kuehl_akt" shows the setting of the cooling

So I guess the only nice way to reproduce this in HA is either to show these 3 sensors or use a template sensor in order to show just one state (at least heating or cooling)

Does that answer your question @remyderuysscher ?

remyderuysscher, 2020-06-30:
@Bouni I had a quick look but for example ID_Ba_Bw_akt is already defined in the service, so how can this show the temperatur of the hot water?

Bouni, 2020-06-30:
In my opinion **ID_Ba_Bw_akt** is the short form of the german ID_**B**etriebs**a**rt_**B**rauch**w**asser_**akt**uell, which means roughly translated ID_Operation_mode_hot_water_current.

So this gives you just the mode for the hot water. In order to get the current hot water temparature you'll need to use **ID_WEB_Temperatur_TBW**

remyderuysscher, 2020-06-30:
> In my opinion **ID_Ba_Bw_akt** is the short form of the german ID_**B**etriebs**a**rt_**B**rauch**w**asser_**akt**uell, which means roughly translated ID_Operation_mode_hot_water_current.
>
> So this [...]

I was just reffering to your comment:  "Parameter 4 "ID_Ba_Bw_akt" seems to show the setting of the hot water"

Bouni, 2020-06-30:
Maybe setting was the wrong word and mode is more accurate then.
Can I close this issue?

remyderuysscher, 2020-07-01:
> 
> 
> Maybe setting was the wrong word and mode is more accurate then.
> Can I close this issue?

Sorry really busy, I'll let you know asap if I find the correct status with the pull request from WhistleMaster.

WhistleMaster, 2020-07-03:
>     * Parameter 108 "ID_Einst_BA_Kuehl_akt" shows the setting of the cooling

I did not find this setting on your [loxwiki](https://www.loxwiki.eu/pages/viewpage.action?pageId=18219334). I'll try it.

WhistleMaster, 2020-07-03:
I guess the problem is that `ID_WEB_WP_BZ_akt` does not report the "cooling" status. All other status "heating", "water", "no request" are correct, only "cooling" is missing. Seems to be a firmware issue. I'll try to update again to V.3.85.5 to see if it fixes that.

Bouni, 2020-07-03:
I think that AlphaInnotec initially had no cooling and they added it later on an thus did not represent it in `ID_WEB_WP_BZ_akt`, but that's just a guess.

remyderuysscher, 2020-07-04:
> I looked a bit closer into this and duped all data from my heatpump 3 times with different settings as displayed in the AlphaApp (Android App):
>
>     1. Heating off, hot water automatic, cooling [...]

Hi @Bouni how can I read out those parameters? These are not calculations?

Bouni, 2020-07-06:
Hi @remyderuysscher 

simply by using group parameters instead of calculations:

```
sensor:
  - platform: luxtronik
    sensors:
      - group: parameters
        id: ID_Einst_BA_Kuehl_akt
```

remyderuysscher, 2020-07-06:
> 
> 
> Hi @remyderuysscher
> 
> simply by using group parameters instead of calculations:
> 
> ```
> sensor:
>   - platform: luxtronik
>     sensors:
>       - group: parameters
>         id: ID_Einst_BA_Kuehl_akt
> ```

@WhistleMaster @Bouni  

Roger. ID_Einst_BA_Kuehl_akt and ID_Ba_Hz_MK3_akt show also automatic with my model.  Unknown_Parameter_1123 shows 0 and 1124 shows 50.

remyderuysscher, 2020-07-06:
Closed as this is a bug in the firmware, hopefully they will fix it in a future release.

closed 2020-07-06 by remyderuysscher