Bouni/python-luxtronik #18: Negative Timestamps

jaxer, 2020-09-08:
Hello,

Thanks for this component.
I just got a new heatpump. Only couple errors on it, so out of 5 slots - 3 are empty.
When loading this component i get errors about invalid parameter on this line:

https://github.com/Bouni/python-luxtronik/blob/910ab0ed894cb8e3d686d812992e38055286907c/luxtronik/datatypes.py#L120

After debugging turned out that heatpump returning negative number for value when error is not there. I modified code from:
```
        return datetime.datetime.fromtimestamp(value)
```
to
```
        if value > 0:
            return datetime.datetime.fromtimestamp(value)
        else:            
            return datetime.datetime.fromtimestamp(0)
```

after which i was able to initialize library and read value from heatpump:

```
>>> l.calculations.get("ID_WEB_ERROR_Time0")    
2020-06-02 15:20:33
>>> l.calculations.get("ID_WEB_ERROR_Time1") 
2020-06-02 14:39:52
>>> l.calculations.get("ID_WEB_ERROR_Time2") 
1970-01-09 01:51:28
>>> l.calculations.get("ID_WEB_ERROR_Time3")  <--- it is failing here in master
1970-01-01 02:00:00  
>>> l.calculations.get("ID_WEB_ERROR_Time4") <--- it is failing here in master
1970-01-01 02:00:00
```

Probably there are some other booleans somewhere that say if error is present or not, and if its not then timestamp value is undefined and in my case it was some random negative number.. ?

Bouni, 2020-09-11:
Hi,
I'll take a look into this issue the next few days!

Bouni, 2021-04-07:
Finaly found the time to fix this, sorry for the delay 😓

closed 2021-04-07 by Bouni