Bouni/python-luxtronik #27: Errorcode To localized text

macrojames, 2022-02-18:
Hi,

I had errors today on my heatpump and I wondered if there are readings for it.
I found the time and the number as errorcode and use that for now.

Then I had the idea to add the texts according to the numbers, which will be a local mapping at the moment.
I got the meanings from https://www.alpha-innotec.ch/fileadmin/content/downloads/Fehlerdiagnose_Codes_HMD2_de.pdf
and translated it to csv.
[errorcodes.csv](https://github.com/Bouni/python-luxtronik/files/8099956/errorcodes.csv)

Maybe you want to use that for localized number2text conversions?

Bouni, 2022-03-25:
Hi,

sorry for the late feedback!

I totally agree that having error texts rather than just plain numbers would be great!
But I defenitely want to have them as translations so that we can support multiple languages.
To be honest, I've never looked into the way the Home Assistant folks do translations in integrations.
It would be nice if we can do it th right way directly from the start!

https://developers.home-assistant.io/docs/internationalization/custom_integration/

Would you dig into this and create a PR?

Bouni, 2022-03-25:
🤦🏽‍♂️ Oh man, wrong repo for HA translations. I think we should do the translation over in the [HA integration repo](https://github.com/Bouni/luxtronik)

WolfsWelpe, 2022-04-24:
Hey this is exactly what I'm looking for at the moment!
Is the translation already implemented for the error codes? :D

macrojames, 2022-04-24:
I could not look into this. I tried implementing the localalized json files on another project some time ago but I failed. I can't invest the time right now to dig deeper into this, sorry.

Bouni, 2022-04-26:
I defenitely will take a look at it but that will take some time, my backlog is getting bigger and bigger every week 🙈

kbabioch, 2022-05-05:
Is the idea to have those error codes translated in the core Python library (i.e. here), or in the Home Assistant integration?

Bouni, 2022-05-09:
@kbabioch Actually I'm not sure where the best place is to have translations. If you have an opinion, feel free to share it with me as I've no experiance so far with translations

BenPru, 2023-01-11:
@kbabioch Currently I'm extending my ha integration ([luxtronik2](https://github.com/BenPru/luxtronik)) with a "sensor.luxtronik2_switchoff_reason" and "sensor.luxtronik2_error_reason".
The plan is to use the ha translation to solve the error code into the error description.

kbabioch, 2023-07-11:
I think that we should provide this functionality in this repository. Home Assistant can then reuse or reimplement it, but core functionality should not be handled somewhere.