Bouni/python-luxtronik #32: Password required for Websocket

SchumacherFM, 2022-11-13:
Hey,

I'm trying to connect via https://github.com/BenPru/luxtronik to my heat pump AlphaInnotec LWCV 82R1/3 with the latest Luxtronik Software. But unfortunately the websocket requires a password send as `LOGIN;0000` where 0000 is the password.
AFAIK I cannot remove the password from the websocket .... maybe you know ... on the other hand, can you implement please to send optinoally a password? (I can't code in python :-( )

Thanks!

Bouni, 2022-11-16:
Hi,

`python-luxtronik` does not use websockets. It uses the TCP protocol on port 8889.

You can use nmap to check if your heatpump listens on that port: `nmap -p 8889 <ipaddress>` where <ipaddress> is the IP of your heatpump.

SchumacherFM, 2022-11-16:
Thanks for replying. Your're right. After I read the python docu.

```
$ nmap -p 8889 192.168.yyyy.xxx
Starting Nmap 7.93 ( https://nmap.org ) at 2022-11-16 08:57 CET
Nmap scan report for 192.168.yyyy.xxx
Host is up (0.0012s latency).

PORT     STATE SERVICE
8889/tcp open  ddi-tcp-2

Nmap done: 1 IP address (1 host up) scanned in 0.12 seconds
```

Should be another problem. I'll close this ticket and maybe open another one.

closed 2022-11-16 by SchumacherFM