Bouni/python-luxtronik #33: Add support for Type: Time

rhammen, 2022-11-22:
There are a lot of parameters that have "_zeit_" in their description.
These parameters indicate a time. The value is stored as "seconds past midnight".
So e.g. 
        0 = 00:00:00 (midnight)
  3600 = 01:00:00 (1 am)
23400 = 06:30:00 (6:30 am)
43800 = 12:10:00 
72000 = 20:00:00 (8 pm)
Would it be possible to add support for this?

Bouni, 2022-11-24:
Can you give a concrete parameter that has this format so that I can check it at my pump?

rhammen, 2022-11-24:
This is about paramater with "zeit" in their description.
So for example all parameter with numbers 162-221, 223-282, 284-343, 345-404, 406-505, 507-606, 608-667.
It concerns the times you block e.g. heating or domestic water.
Example:
On your Luxtronik go to Domestic Water --> Clockprogram (not sure of the exact name, since mine is in Dutch)
Then choose 5+2, so weekdays and weekend.
You get to a screen showing "Monday-Friday" and 5 rows, each rows showing a START and END 
My first line reads: 00:00 - 12:10
These values of this first line end up in parameter 416-417. 
So my parameter 416 =0 (00:00) and parameter 417 = 43800 (12:10 = 12*3600+10*60)

If you scroll down the screen past row 5, you get to the screen "Saturday-Sunday".
My first line here reads 00:00 - 14:00.
These values of this first line end up in parameter 426-427.
So parameter 427 = 50400 (14:00 = 14*3600)
 
[image]
[image]

kbabioch, 2023-01-15:
I'm happy to work on this, although I'm rather unsure about how to best handle this within the code.

A parameter value is always 8 bytes long and cannot hold all the information that is mentioned above. That's the reason why it is spread across multiple parameter entries. This is fine in general, but so far the code and datatypes work with the assumption that conversions and representation can be done based on a single parameter.

Any ideas on how to best adjust the code here @Bouni and/or @benpru? Essentially we need a data type / structure that spreads across multiple parameters. Whenever something is changed, potentially multiple parameters need to be reset.

gerw, 2023-01-15:
Are you really sure that a parameter is 8 byte long? I would guess that it is 1 word = 4 byte = 32 bit.

And each time (e.g. "0:00" or "14:00") is still in a single parameter.

kbabioch, 2023-01-15:
> Are you really sure that a parameter is 8 byte long? I would guess that it is 1 word = 4 byte = 32 bit.

At least this is how data is represented internally.

The de-compiled version of the `getParameter` function looks something like this:

```
bool getParameter(eParam paramter_id,int *parameter_value)

{
  bool found_parameter;
  
  found_parameter = false;
  if (((-1 < (int)paramter_id) && ((int)paramter_id < 0x477)) &&
     ((&ParamCollection)[paramter_id * 8] != '\0')) {
    *parameter_value = *(int *)(&DAT_00735a08 + paramter_id * 8);
    found_parameter = true;
  }
  return found_parameter;
}
```

> And each time (e.g. "0:00" or "14:00") is still in a single parameter.

That's true. So probably it's good enough to have a datatype for the parameter. Another module might be needed later on top of `parameters` to combine multiple parameters into a single data structure.

Bouni, 2024-02-09:
This is resolved by #147