Bouni/python-luxtronik #36: Optimizing data request

BenPru, 2023-01-11:
I have some challenges to make my ha luxtronik integration stable:

- Not all write commands arrive at the heatpump. [#54](https://github.com/BenPru/luxtronik/issues/54)
- Written values have a delay. [#48](https://github.com/BenPru/luxtronik/issues/48)
- V2.88.1 firmware crashes when requesting or writing many times in a short period of time.

I have figured out that the firmware has a problem with the reconnects on the socket.
So I have made some changes in the core module:

- Added a thread lock to only run one command (read/write) at a time.
- Close socket connection only in the desctructor.
- Read the values after write command to get the changes promptly.

At the moment I'm still testing, but you might want to take a look.

kbabioch, 2023-01-11:
To my understanding @BenPru is working on this currently in: https://github.com/BenPru/python-luxtronik/tree/Optimizing-data-reques

BenPru, 2023-01-16:
@kbabioch I have fixed the linter messages. But the workflow is using the old source.
https://github.com/Bouni/python-luxtronik/actions/runs/3931922636/jobs/6724000633#step:5:1
How can I force a re run with the current source?

kbabioch, 2023-01-16:
Is this still an issue or have you fixed it by now? The merge request #55 looks fine as far as I can tell in terms of linting?

Bouni, 2023-01-17:
@kbabioch Did you test the changes in #55 ? I cannot test it at the moment ...

BenPru, 2023-01-18:
After the merge it is fixed for me.

PR #55 (2023-01-18): Optimize data requests
Resolves #36
closed 2023-01-18 by BenPru