Bouni/python-luxtronik #44: Additional maintainers

kbabioch, 2023-01-15:
As seen in #40 having only one maintainer can be an issue when this person is not available. Since there is some interest and there are "core contributors" that are familiar with the code base I suggest to create a "maintainer team" that is able to perform organizational tasks in this repository (merge, review, etc.).

Since I've worked with the code base in the last couple of days, I'm willing to take up this responsibility here. I also suggest that @benpru is considered, so that we have at least three persons that can actively work in this repository and manage incoming pull requests, etc.

We could also establish an approval process, so that no change is "just merged", but needs to be approved by at least two persons.

Bouni, 2023-01-16:
Holy smokes, this was a wild weekand judging by the number of new issues and discussions going on :star_struck:

@kbabioch @BenPru I defenitely concider getting you guys in as maintainers, I only have to chek how to do that 😅 
Never had a repo with more than me as maintainer.

@kbabioch Do you know how to set up such a "need two approvals" thing? I know that there's a lgtm bot of some sort, so if two maintainers comment lgtm that enables the merge button

kbabioch, 2023-01-16:
> @kbabioch Do you know how to set up such a "need two approvals" thing? I know that there's a lgtm bot of some sort, so if two maintainers comment lgtm that enables the merge button

I'm more familiar with GitLab than with GitHub, but it should be pretty similar.

Basically these are the steps that need to be done (as far as I can tell):

- Protect the `main` branch
- Require reviews
  - Either set up a number of required reviews (https://github.blog/2018-03-23-require-multiple-reviewers/). To my understanding everyone can review then, though.
  - Set up a CODEOWNERS file that describes who is responsible and allowed to review

If you want, we can discuss the details also via some other channel (mail, preferably chat), so that we can configure and test it.

Bouni, 2023-01-16:
Awesome, I'll look into it in my lunch break!
Maybe we can have a chat in Matrix or your pefered chat platform!

Bouni, 2023-01-16:
I've added @kbabioch and @BenPru as collaborators and resticted review to collaborators. Branch protection is still not clear and need to be checked

kbabioch, 2023-01-16:
Wow, that was fast :-).  From my point of view seems to work. At least I can see some additional options now (such as editing labels, etc.).

From my point of view we could close this issue, once @benpru also verified that it works for him.

If further need to change something about it comes up in the future, we could create new issues for it.

Bouni, 2023-01-16:
👍🏽

Bouni, 2023-01-17:
@kbabioch @BenPru Can you rebase and merge now?

kbabioch, 2023-01-17:
Yes, I can see the merge button. Thank you.

Bouni, 2023-01-17:
Ok, I'll let this issue open until we managed to protect the main branch and establish the review process.

BenPru, 2023-01-18:
> Can you rebase and merge now?

Yes.

Bouni, 2023-01-19:
@kbabioch @BenPru 

I've protected the main branch
- Require a pull request before merging 
  - Require approvals: 2
And
- Require status checks to pass before merging 
  - Require branches to be up to date before merging 

I hope that works as I expect 😅

kbabioch, 2023-01-19:
> I've protected the main branch

Cool.

> Require status checks to pass before merging
>
>    Require branches to be up to date before merging

Also very good.

> Require a pull request before merging
>
 >   Require approvals: 2

Not so sure about the 2 approvals. Sounds good in theory, but essentially right now it means that we need all of us to be available to move on, i.e. when you create a pull request, it takes me and @BenPru to approve it. If anyone of us is unavailable, we're stuck.

Maybe we should start with 1 here and increase it once we have 1-2 more maintainers.

Or we accept that it will take some time for pull requests to be approved :-).

Bouni, 2023-01-20:
You're absolutely right, I wasn't talking into account that the author of a PR cannot review 😅
I'll change it to 1 review tomorrow