Bouni/python-luxtronik #46: Provide raw data

BenPru, 2023-01-15:
Currently it is not possible to get the luxtronik raw value. The most values are automaticly converted.
This is good in the most cases but for example it is not possible to get a raw export.
I have added a ha diagnostic for my integration which exports a dump of all luxtronik values. But I would like to add a field for the raw values.

kbabioch, 2023-01-15:
@benpru Where do you think this needs to be implemented?

BenPru, 2023-01-16:
- For raw data in the dump.
- Some senors have no type implementation and the data is provided raw. Other sensors have a type implementation and translate the enum value to a text. See SwitchoffFile vs. Errorcode.

Bouni, 2023-01-16:
I think '.value' of every class should give you the raw value as far as I remember

BenPru, 2023-01-16:
> I think `.value` of every class should give you the raw value as far as I remember

Unfortunately not, it presents the converted value returned by `from_heatpump`.
https://github.com/Bouni/python-luxtronik/blob/main/luxtronik/parameters.py#L1167

Bouni, 2023-01-16:
You're right, but adding raw that holds the raw value is very easy

kbabioch, 2023-01-16:
@Bouni I think this might be related to #49.

From my understanding the `value` attribute is never used.

> You're right, but adding raw that holds the raw value is very easy

We could add another attribute `raw_value`, which gets filled whenever something is read from the heat pump.

Guzz-T, 2023-05-01:
Do the changes in #92 fix this issue too?

gerw, 2023-07-07:
@Guzz-T I think so.

kbabioch, 2023-07-11:
> Do the changes in #92 fix this issue too?

Yes, the raw value can now be accessed using the `raw` property, for instance:

> >>> a.get(81)
V3.85.8
>>> a.get(81).raw
[86, 51, 46, 56, 53, 46, 56, 0, 0]

So, values are being converted / represented by the internal datatype (if available), but the raw data can also be accessed.

I'm closing the issue to keep the backlog small-ish, feel free to re-open it (or create another one), if something essential is missing.

closed 2023-07-11 by kbabioch