Bouni/python-luxtronik #53: Add discover function

BenPru, 2023-01-16:
I have extend bounis discover function and implement it in my integration:
https://github.com/BenPru/luxtronik/blob/3912b8e8219d4c93101ac71df542beeb60b5ed59/custom_components/luxtronik/helpers/lux_helper.py#L11

But I think it should be in the core python luxtronik.
It should discover not only one heatpump in case of more than one. It should return a list with ip or name and port.

Bouni, 2023-01-18:
You're absolutely right, this should be part of this module rather than the HA integration!
Should be not too hard to move into here, right?

kbabioch, 2023-01-18:
Having the functionality in the library sounds good. On the other hand I'm not quite sure yet where it would be ideally placed. Probably we need also some kind of helper functions that could be invoked prior to connecting to the controller itself via the `Luxtronik` class.

Bouni, 2023-01-18:
Maybe if no ip / port is passed when the class is instanciated, it tries to autodiscover the heatpump.
The only questions how it would handle a situation where more then one heatpump is in the same network 🤔

kbabioch, 2023-01-18:
> Maybe if no ip / port is passed when the class is instanciated, it tries to autodiscover the heatpump.

Personally I'm not in favor of too much "magic" in a Python library. Having a help function that will detect heat pumps and return a list of IPs would be good enough. Automatic connections should be handled by other solutions such as Home Assistant, etc.

As you've said there are some considerations when there are multiple heat pumps in the network for example.

Also it should not be too easy to use the library. What I mean by that is, that (at least to my understanding) there should be some explicit act of commitment, e.g. like specifying an IP and invoking some kind of initialization. If we make it too easy, people could import a module that they don't know yet, and already connect to their heat pump - potentially doing something stupid.

What I think might be a good solution:

- Have functionality to detect heat pumps automatically in the module (some kind of help function). This function will return a list of IPs / ports to connect to.
- Manual initialization (like now)
- Additionally (far future vision): Have a command-line utility (or even GUI) that interacts with the heat pump, i.e. you can interact with all kinds of values from the heat pump via the CLI utility. In this utility we could implement auto-connect features.

BenPru, 2023-01-18:
Many people install this package manually to get a dump of all values or to find the heatpump.
Why don't we make it as easy as possible for them?
If no param is commited just discover all heatpump and print the ip/name with the port and then a dump of all params and calculations. The visibilities are not relevant I think.
If we detect more than one heatpump just dump the ips and ports.
A second call with the ip can print dump.

Bouni, 2023-01-18:
In my opinion we should, as @kbabioch suggested, have the ability to autodiscover, but that should be by intent.

When using this module standalone, we should provide a nice CLI, something like `luxtronik autodiscover` which gives you a list of heatpumps with their IP/port.

From there on the user can use `luxtronik --ip ... --port ... dump` for example.

BenPru, 2023-01-18:
For example see [onkyo-eiscp](https://github.com/miracle2k/onkyo-eiscp)
`onkyo --discover` --> discover
`onkyo --host 172.20.0.144 volume=55` --> port is optional because its always the same. 8889 as default for our case?
`onkyo -a ...` --> choose discovered device for the following command.

Bouni, 2023-01-20:
I've copy and pasted @BenPru s discover function into a seperate file.
That way it can be used in his HA integration for example with simply importing it `from luxtronik.discover import discover` and like the one in his inegration.
I fixed several linter errors, so its not a real 1:1 copy.

closed 2023-01-21 by Bouni