Bouni/python-luxtronik #54: Implement unit testing

kbabioch, 2023-01-16:
Currently there are not automatic test cases for the code base. This could be improved by usage of pytest in order to allow for automated tests in the pipeline.

kbabioch, 2023-01-16:
Currently working on this in: https://github.com/kbabioch/python-luxtronik/tree/pytest

kbabioch, 2023-01-16:
I have a basic proof of concept in place at https://github.com/kbabioch/python-luxtronik/tree/pytest. I will re-organize the tests a little bit.

Is there any feedback or things that you need or expect from this @Bouni @BenPru?

Things that I will hopefully add later:

- More extensive test cases for network connectivity, etc.
- Code coverage reports that will be integrated into GitHub
- Re-organization of the test code into classes
- Usage of fixtures and more data sets

Bouni, 2023-01-18:
I just had a look at your feature branch and the PR there (https://github.com/kbabioch/python-luxtronik/pull/2)
That looks awesome 😍

kbabioch, 2023-01-18:
Well, the concept itself seems to work fine, currently working on more unit tests for all of the available datatypes. This also lead me to discover some inconsistencies.

Also I'm wondering: How strict should we be with the datatypes. Right now they are implemented in a lenient way, e.g. it converts typical values and works fine for most purposes. However there are some issues and edge cases that are not covered.

Is it worthwhile to consider this (for later), or do we want to stick to the "easy" approach.

Some examples/issues/things to consider:

- Should it be possible to convert values other than integers for datatypes such as `Celsius` or `Kelvin`? I guess the heat pump will only deal with integer values, but the function itself also converts anything else.

This will raise an `ValueError`:

```python
Celsius("").from_heatpump("")
```

This will raise an `TypeError`:

```python
            assert Celsius("").from_heatpump("")
```

Also, in some cases converting back and forth, will trigger inconsistencies:

```
>>> luxtronik.datatypes.Celsius("").from_heatpump(luxtronik.datatypes.Celsius("").to_heatpump(0.11))
0.1
```

Any opinion on what to do about this? Keep the tests simple and ignore this, or make the code more complex by catching / dealing with all sorts of special cases :-).

Bouni, 2023-07-20:
@kbabioch Why did we comment out [this section](https://github.com/Bouni/python-luxtronik/blob/main/.github/workflows/pytest.yml#L24-L28) of the test action?

I would really love to have coverage reports 🤩

kbabioch, 2023-07-20:
As far as I remember there has been an issue with permissions. Let's try to enable it again and address the issues.

gerw, 2024-02-22:
I am closing this issue, since unit tests are running successfully.

closed 2024-02-22 by gerw