Bouni/python-luxtronik #57: measurement_type is used inconsistently

kbabioch, 2023-01-17:
It seems like the `measurement_type` is used inconsistently throughout all of the different datatypes. In some cases it is used to describe the type of data (e.g. `boolean`), in some cases as unit (`celsius`) and in some other cases as abbreviation of the unit (`Hz`).

For instance here it is used as unit (`celsius` written out fully):

https://github.com/Bouni/python-luxtronik/blob/fec29e97a3c0bfd5453bc5af69854d1a6a716009/luxtronik/datatypes.py#L53-L56

In case of `Frequency` it is used as unit (`Hz`, which is an abbreviation for `Hertz`):

https://github.com/Bouni/python-luxtronik/blob/fec29e97a3c0bfd5453bc5af69854d1a6a716009/luxtronik/datatypes.py#L77-L80

In case of other datatypes, it more used as a generic description of the type of data that is being represented, e.g. `boolean`:

I would suggest to make this more consistent by doing the following:

- Rename `measurement_type` attribute into `data_type` (since its not always a measurement)
- Introduce a new attribute (`unit`) to describe the unit of the measurement (e.g. `Hertz`, `Watt`, `Kelvin, `Celsius`, etc.).
- (Unsure if needed at this point: Another attribute that contains a visible representation of the `unit`, e.g. `Hz, `W`, `K`, `°C`)

Bouni, 2023-01-17:
This makes total sense! 

Do you see a chance to kind of auto match HA SensorDeviceClass and Unit in the integration if we act foresighted?

kbabioch, 2023-01-18:
> Do you see a chance to kind of auto match HA SensorDeviceClass and Unit in the integration if we act foresighted?

Yes, that is what I had in mind here. We could use the `type` to represent / map to [HomeAssistant's Sensor device classes](https://developers.home-assistant.io/docs/core/entity/sensor/#available-device-classes).

The only problem that I see is that we're currently mixing sensor values (e.g. things that you measure like the temperature) together with some logical values (e.g. boolean), which are `BinarySensors` in Home Assistant.

Probably it should be easy to map within the Home Assistant integration, when we stay consistent and close to the Home Assistant values in our code here.

kbabioch, 2023-01-30:
So, any further additions / suggestions / improvements? Otherwise this might be worked on once #54 is merged.

Bouni, 2023-01-30:
I approved #54 a few minutes ago, so you're good to go with this one!

kbabioch, 2023-07-11:
While #49 has been closed, it probably contains the best summary / data representation so far, including the changes proposed here:

https://github.com/Bouni/python-luxtronik/issues/49#issuecomment-1385508468

closed 2023-07-12 by kbabioch