Bouni/python-luxtronik #63: Handling of big values in IPAddress datatype

kbabioch, 2023-01-19:
While working on unit tests I've noticed that the `IPAddress` datatype is trying to handle large integer values (> `2**32`) in the following way:

https://github.com/Bouni/python-luxtronik/blob/973f7c98167fa837f316e63d73aecebf5a2f1785/luxtronik/datatypes.py#L95-L111

As you can see the `from_heatpump()` tries to deal with values smaller than 0 and larger than `2**32`. This leaves me wondering:

- Is the heat pump really returning such values, or is this some kind of precaution in case it would?
- Are the calculated values representing the correct IP addresses then? At least the `ipaddress` library considers anything between `0` and `2**32` as IPv4 addresses, but anything outside of this range not.

In the `to_heatpump()` function once again the case of `2**32` is considered. However I don't get how this could ever be happening. Actually the `ipaddress` will raise an error:

> ipaddress.AddressValueError: 14294967040 (>= 2**32) is not permitted as an IPv4 address

Therefore this case cannot occur and it doesn't need to be handled?

Suggestion/Question: Can we drop those checks in favor of a simpler datatype, or are they needed?

Bouni, 2023-01-19:
Apparently there was an issue with IP Adresses < 128.0.0.1 (See #8 )

Bouni, 2023-01-23:
I did a few test with my heatpump's IP adresses.

For parameter 750, ID_AdresseIP_akt my heatpump reports a value of `-1062709237` which must translate into `192.168.88.11`.
Using the ipaddress packet this fails:

```python 
ipaddress.ip_address(-1062709237)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.10/ipaddress.py", line 54, in ip_address
    raise ValueError(f'{address!r} does not appear to be an IPv4 or IPv6 address')
ValueError: -1062709237 does not appear to be an IPv4 or IPv6 address
```

Adding 2^32 to that value gives the right ip address:

```python
ipaddress.ip_address(2**32 + -1062709237 )
IPv4Address('192.168.88.11')
```

Also using module works:

```python
ipaddress.ip_address( -1062709237 % 2**32)
IPv4Address('192.168.88.11')
``` 

The bigger "problem" is the writing part

```python
2**32 - int(ipaddress.ip_address('192.168.88.11'))
1062709247 # wrong value
```

```python
2**32 % int(ipaddress.ip_address('192.168.88.11'))
1062709247 # wrong value
```

So I searched a bit further and found that this can be nicely down using `socket.inet_ntoa` + `struct.pack` / `socket.inet_aton` + `struckt.unpack`

```python
socket.inet_ntoa(struct.pack('>i', -1062709237))
'192.168.88.11'
```

```python
struct.unpack('>i', socket.inet_aton('192.168.88.11'))[0]
-1062709237
```

That is in my opinion by far the superiour solution for this! 

@kbabioch what do you think about this?

kbabioch, 2023-01-24:
@Bouni Cool findings. I'm currently trying to understand how this data is handled internally to make sure we're not missing anything here.

One finding where those parameters are referenced:

After some de-compiling the `setNetworkConfig` function looks like this:

```
void setNetworkConfig(NetworkAddresses_t *network_configuration)

{
  int *piVar1;
  uint32_t ip;
  uint32_t subnet;
  uint32_t broadcast;
  code *pcVar2;
  uint local_30 [2];
  uint32_t local_28;
  FILE *file_handler;
  int i;
  uint local_1c;
  
  piVar1 = appl;
  pcVar2 = *(code **)(*appl + 0x2c);
  ip = htonl(network_configuration->ip);
  subnet = htonl(network_configuration->subnet);
  broadcast = htonl(network_configuration->broadcast);
  (*pcVar2)(piVar1,ip,subnet,broadcast);
  SetNetworkGateway(network_configuration->gateway);
  file_handler = fopen("/etc/resolv.conf","wb");
  if (file_handler != (FILE *)0x0) {
    local_30[0] = network_configuration->field4_0x10;
    local_30[1] = network_configuration->field5_0x14;
    local_28 = network_configuration->gateway;
    for (i = 0; i < 3; i = i + 1) {
      local_1c = local_30[i];
      fprintf(file_handler,"nameserver %d.%d.%d.%d\n",local_1c >> 24,local_1c >> 16 & 0xff,local_1c >> 8 & 0xff,local_1c & 0xff);
    }
    fclose(file_handler);
  }
  storeIPSettings(network_configuration);
  return;
}
```

Unfortunately the pointer to `appl + 0x2c` is unclear to me at this point in time. It's some pointer to the code, but not sure how to know where it is pointing at, to better understand how it is handled then.

At this point in time it looks like the only functions that are being applied is `htonl`.

Maybe @gerw happens to know more about this, since he is also interested in reversing the binary :-).

closed 2023-01-24 by Bouni

gerw, 2023-01-25:
I think that `appl` is set in `CtrlInterface::CtrlInterface`. Thus, `appl + 0x2c` should point to the method `CApplication::setIpAddr`. However, it is not clear to me whath this method is doing.