Bouni/python-luxtronik #68: Consider usage of data classes for datatype

kbabioch, 2023-01-19:
Python's built-in [data classes](https://docs.python.org/3/library/dataclasses.html) are a nice and native way to deal with classes that represent and handle data. In our case we could use data classes for `datatype`. The main advantage is that some methods (such as `__repr__` and `eq`) are already implemented out of the box and you can mainly focus on the name / type of data without too much boilerplate code.

While this is not a high priority, it could still be an improvement to consider for the future.

I would suggest to first implement unit tests, though (e.g. via #64) in order to be able to refactor the code with reduced risk of regressions and issues.

kbabioch, 2023-01-19:
Could also be combined with #49 while re-doing the `datatype` module anyway.

Bouni, 2023-01-20:
Never used dataclasses before but heared of them. They look like the right thing to use in our case as far as I can tell.
We should somehow com up with a plan what to do first becasue it seems that multiple parallel PRs will end up in a rebase fight.

@kbabioch what strategy seems smart in your eyes?

kbabioch, 2023-01-20:
> We should somehow com up with a plan what to do first becasue it seems that multiple parallel PRs will end up in a rebase fight.

Yes, on the other hand we won't do everything at once anyway ;-).

> @kbabioch what strategy seems smart in your eyes?

My current agenda is to implement unit tests for the current implementation (see #54). Afterwards we can start to refactor the code and improve it considering all the discussions we've had so far regarding the datatypes.

Essentially there is enough work left to do, let's see how far we get.

Bouni, 2023-01-20:
Sounds good to me!

kbabioch, 2023-07-11:
After reading up on data classes some more, I don't think it offers any benefit in our case, since we have some custom code already implemented. #113 implements further functions (repr, eq, lt, etc.), so that the usage of data class doesn't make any sense for now.

kbabioch, 2023-07-11:
This can be closed once #113 is merged.

closed 2023-07-12 by kbabioch