Bouni/python-luxtronik #75: Add support for Luxtronik1 controllers

Bouni, 2023-01-24:
I have a request of someone that has a heatpump with a Luxtronik1 controller if we could support these as well.

The protocoll is completely different, here's a good overview: https://github.com/christophgysin/heatpumpctl/blob/master/doc/AlphaInnotec_SiemensNovelan_Luxtronic_V24.png

I first thought creating a seperate python module for it but then there will be the name problem. This module schould have been named `python-luxtronik2` from the start on. Then I could simply name the other one `python-luxtronik1`.

What do you guys think about integration support for Luxtronik1 controllers into this module?

We could rename the [main class](https://github.com/Bouni/python-luxtronik/blob/main/luxtronik/__init__.py#L45) from `Luxtronik` to `Luxtronik2` and have a `Luxtronik1` class for the other controllers ....

kbabioch, 2023-01-24:
Hm, I can totally understand the "feature request" for support of this controller.

Looking at the screenshot, it looks like the protocol for `luxtronik1` is a plain text protocol, so there won't be much in common with our existing codebase.

We could, of course, deal with this in the code and support both controller versions. But technically, I would prefer two modules, as this makes it easier to keep everything leightweight and well documented. Also keeping track of bugs, etc. will be more complicated, as you always need to ask / ensure which version is involved.

As you've said in a perfect world there should be one module called `luxtronik` and another one called `luxtronik2`. Unfortunately our current situation is not ideal :-).

Not sure, how much pain would we create, when we follow the following strategy:

- Rename this repository to `python-luxtronik2`
- Create a new repository called `python-luxtronik` to use for development of a new library
- Add information to the `README` file of both repositories
- Publish the code of `python-luxtronik2` as `luxtronik` and `luxtronik2` on `test.pypi.org` and `pypi.org` (with README information)
- Once there is enough progress on the code for the version 1 controller:
  - Publish the code of `python-luxtronik` as `luxtronik` on `test.pypi.org` and `pypi.org`
  - Publish the code of `python-luxtronik2` as `luxtronik2` on `test.pypi.org` and `pypi.org`
  - Replace README information with references to the other repository / project and keep a warning that this code will only work for version `X` of the controller

While this is somewhat painful, it's probably still better to do it now then later, when there might be more user and a bigger user base (Home Assistant, etc.).

Or is renaming this, totally out of question?

Bouni, 2023-01-24:
I think its better to create a seperate module after a bit of thinking about it. I might call it `python-luxtronik-legacy`.
The chance of confusing the two if we name it `python-luxtronik1` is to big in my eyes.

Renaming this one is not ideal in my eyes.

Bouni, 2023-01-24:
I think 2 rfers to the Generation of the Hardware