Bouni/python-luxtronik #77: Introduce pre-commit-hooks

Bouni, 2023-01-25:
I think we should start using pre-commit-hooks as for example HA does!

See https://github.com/home-assistant/core/blob/dev/.pre-commit-config.yaml for reference

That way we could have nice things like **black** formatting, **pylint** or **ruff** for linting, **isort** for nice import order, and **autoflake** to remove unused imports and variables.
That way we can fix issues before we push code and GH actions show that there are improvements necessary.

Optinions?

kbabioch, 2023-01-25:
Yes, having pre-commit hooks might add **additional** convenience. Since it can be disabled / ignored locally, we still need the important checks in in the CI/CD pipeline.

Should we just copy the essential stuff from Home Assistant?

Bouni, 2023-01-25:
Yes, that was my intention. I work on it right now, but I think I went down a deep rabbit hole 🙈 

- setup.py -> legacy, should be replaced with pyproject.toml
- pyproject.toml -> looking at HA's pyproject.toml, jeez what are all these settings 🤯 

A lot to cleanup I guess, maybe we should tackle that in a first PR and after that's done start with the pre-commit-hooks

kbabioch, 2023-01-30:
@Bouni Any help you need here? I guess you're working on this already in the background and have already prepared some changes? Looking forward to those changes ;-).

Bouni, 2023-01-30:
I worked on this last week. The biggest issue is that I need to change many thing in order to get the pre-commit hooks to be statisfied 😆 

I tink I'll add them commented out in the first place, then we can "activate them" one by one and change the necessary files. Otherwise we end up with a giant change set.

One more thing, I would really like to add `black` as the first pre-commit hook becuase that will solve a lot of ruff/pylint warnings/errors because it wraps long lines for example.

Are you ok with that?

kbabioch, 2023-01-30:
Yes, for me that is totally fine and having black as pre-commit hook is also okay.

However I would not like to have automatic rewritting of code as part of the pipeline, i.e. we could run black only in some kind of reporting mode when we integrate it into the pipeline.

The "problem" with pre-commit is, that its more of a "nice-to-have", and not mandatory. So not everyone will execute it when submitting PRs.

Bouni, 2023-01-30:
We could run black with the `--check` flag: https://black.readthedocs.io/en/stable/usage_and_configuration/the_basics.html#writeback-and-reporting

but to be honest, I run black on almost all my projects and it never produced code that didn't work afterwards.
So in my opinion we could run black as a pre-commit hook in write mode, but if you still insist, I'm also fine with `--check`

kbabioch, 2023-01-30:
It's fine to have it in pre-commit. I would argue to not add it in "rewrite mode" in the CI/CD pipeline. I saw projects doing that and committing code to the repository that no one has seen.

Bouni, 2023-01-30:
I did a quick test, and it seems to work with `--check`:

```
black................................................(no files to check)Skipped
[pre-commit d960df1] Adjust regex
 1 file changed, 1 insertion(+), 1 deletion(-)
```

But its rather pointless to run black in read only mode as I think more about it. The sole purpose of black is to take away the need for the programmer to fully format the code nicely.

When having black as a pre commit hook we could achive a contant code format quality level. Then e could add a GH action with black in read only mode so that the check fails if someone submits a PR without the proper black formatting.

Bouni, 2023-01-30:
😆 Oh snap, I was just a few seconds to late with my answer. Ok so I will have black in write mode in the pre-commit-hooks and will supply a PR for CI/CD with a read-only check.

Bouni, 2023-01-30:
Whta do you think of a PR that has black ran over all source files in order to get them to a level from where we can continue?

kbabioch, 2023-01-30:
Also fine with me, should we do it along with #81 or afterwards in a new PR?

Bouni, 2023-01-30:
I think we could do it in the same PR, give me a second

closed 2023-01-30 by Bouni