Bouni/python-luxtronik #85: Support multiple object instances

Guzz-T, 2023-02-27:
The current implementation uses class variables for the fields `Parameters.parameters`, `Calculations.calculations` and `Visibilities.visibilities`. Due to this the memory to save the read data is shared between all instances of the `class Luxtronik`.

#### Possible solution1:
Do not buffer the read data, but return them by calling `read()`. Just define the data structure within the `Calculations.py` (and so on).

#### Possible solution2:
Move the variable declaration into the `__init__()` section of the `class Calculation` (and so on).

In terms of the OOP, solution 1 would fit better, since the data read does not belong to the luxtronik-read/write-interface

Bouni, 2023-02-27:
Interested in submitting a PR? I have very limited time at the moment 😓

Guzz-T, 2023-02-27:
I could give it a try.

Guzz-T, 2023-02-28:
Got a first draft. Could you give me the rights to push a branch?

Guzz-T, 2023-03-01:
Opened a pull-request. See: https://github.com/Bouni/python-luxtronik/pull/86

PR #86 (2023-04-13): Issue-85: support multiple object instances
fixes #85 

Branch tested with home-assistant 2022.9.7 and a master/slave AIT-heatpump setup. Therefore i added a second renamed "luxtronik" integration (see https://github.com/Guzz-T/luxtronik/tree/try/second-integration-for-slave-heatpump).

Pull-Request should not contain any breaking changes. The diff in `calculations` (and ...) seems pretty hard, but I just had to change the indentation (= use 'hide withspaces').
closed 2023-04-13 by Bouni