Bouni/python-luxtronik #98: Following the README leads to an error

haarts, 2023-06-18:
I've diligently followed the README and did a `pip install luxtronik`. That installed the latest version 0.3.14. Then I opened a Python REPL and was greeted with an `TypeError: cannot unpack non-iterable NoneType object` error. The `read()` method on the `l` object returned nothing. The error makes sense. 

On closer inspection, the code I installed with `pip` lagged far behind the code in `main`! Resulting in this unfortunate error. 

Perhaps it is a good idea to release a new version aligning what people install with the README?

Bouni, 2023-06-28:
You're right, but we don't want to release a new version at the moment because there are more breaking changes to come whcih we want to bundle in a 0.4 version or something like that.

I guess we should have dev branch and keep the main branch equal to the latest release 🤔

Bouni, 2023-06-28:
In the meanwhile you can follow the instructions here:

https://github.com/Bouni/python-luxtronik/tree/0.3.14

Or if you prefer the dev version, install it directly from git:

`pip install 'luxtronik@git+https://github.com/Bouni/python-luxtronik.git#main'`