Bouni/python-luxtronik #99: Calculation ID 258

RobinFlikkema, 2023-06-28:
Hi all,

It seems that ID 258 in calculation.py is returning the software version running on the RBE 

For me this is 1.12 and the data returned is "112". My luxtronik controller runs version v3.85.8

Anyone else seeing the same?

kbabioch, 2023-06-28:
What exactly is RBE in this context?

Bouni, 2023-06-28:
I think it's short for "RaumBedienEinheit", a remote control thingy for the heat pump

kbabioch, 2023-06-28:
Ah, okay, don't have it, so can't verify.

Bouni, 2023-06-29:
@RobinFlikkema I do not have a RBE but your guess seems right.
Any chances that you update the RBE Firmware sometime soon so that we get a chance to see if the number changes?

RobinFlikkema, 2023-06-29:
I would try that if I knew how to update that thing.

Maybe someone else has this RBE with a different firmware and can crosscheck?

kbabioch, 2023-07-11:
For me this value is:

> Number: 258   Name: Unknown_Calculation_258                                      Type: Unknown              Value: 0

Probably 0 means "no RBE" then. Are we sure that only value 258 is relevant here, i.e. that the version is represented in only on parameter and not i.e. in multiple ones? What is 259 saying for you?

In my case it is:

> Number: 258   Name: Unknown_Calculation_258                                      Type: Unknown              Value: 0
> Number: 259   Name: Unknown_Calculation_259                                      Type: Unknown              Value: 1

RobinFlikkema, 2023-07-11:
Hi @kbabioch 

See here:
> Value 256: 0
> Value 257: 0
> Value 258: 112
> Value 259: 1

So unsure if there is a connection with 259 for example

kbabioch, 2023-07-11:
Do you have a screenshot from the web interface showing this value, i.e. how exactly it is called?

RobinFlikkema, 2023-07-11:
[image]
It is shown here at the bottom, freely translated "Firmware version RBE"

kbabioch, 2023-07-11:
Okay, I'm just wondering how `112` is converted into `1.12` here. Probably the first byte is the "major" version and the second byte the "minor" version.

But does the web interface really have some logic to convert / understand every kind of calculation?

gerw, 2023-07-11:
The webinterface gets human readable data from the heat pump already. The conversion is done on the heat pump itself.

kbabioch, 2023-07-11:
> The webinterface gets human readable data from the heat pump already. The conversion is done on the heat pump itself.

Ok. Do you happen to know how this functionality is implemented? Is it reversible via Ghidra? Might be an interesting location to have a closer look at :-)

gerw, 2023-07-11:
No, not really. There are many functions `fill_menu` (or similar), which seem to be responsible to fill the menus (surprise!), in particular in the settings of the heat pump (all the lists with temperatures, etc). These menus seem to be mirrored to the web interface itself. How ever, I was not able to understand these functions...

gerw, 2023-07-11:
I guess that 259 is related to cooling, see #112.

gerw, 2023-07-11:
I digged again in the "source" code. Actually the setting of the calculations is done in `WebServerAktualisieren` (aka `refresh web server`). Here, one can find the code which sets calculation 258:
````
  if (Einst_RFVEinb_akt == 4) {
    this = (MB_RMU *)MB_RMU::getInstance();
    local_2c = MB_RMU::getRevision(this);
  }
  else {
    local_2c = 0;
  }
````
but
````
undefined2 __thiscall MB_RMU::getRevision(MB_RMU *this)

{
  return *(undefined2 *)(this + 0x42);
}
````
is not quite helpful (I think that it returns a member variable).

kbabioch, 2023-07-11:
It seems to be trivial to calculate `1.12` from `112`, but I would still prefer to understand what exactly is done here (and where). Not sure, if we will find out, though :-).

kbabioch, 2023-07-13:
I've worked on this in #118. If possible, would be great to have some additional feedback. Also having other values reported from the "real world" might be interesting, since we only have two data points so far.

closed 2023-07-14 by kbabioch