Bouni/python-luxtronik #100: New parameters

michischatz, 2023-07-02:
Hi,

I got a new update on my Luxtronik (manufacturer "Alpha Innotec")  with software version 3.88.0 (I saw 3.88.1 is alread released for my installed type). I also used the old version of python-luxtronik which was working fine with my old firmware of the heat pump controller. After the update I got warnings and I checkt this github rep and recognized, that unknown parameters are not warnings anymore, just debug messages.
The new params are:
Parameters: 1126 - 1142
Calculations: 260 - 266
Visibilities: 355 - 365

Maybe we could add the new parameters to the libraray. I also noticed a new section of information in my web ui, called "Energy used". Attached you will find an example (unfortunatly screenshot is in German):
[image]

For now I dont know the explanation of the new parameters but I assume it could be the "Energy used" information which are very helpful for tracking the energy usage of the heat pump controller.

Best regards,
Michael

kbabioch, 2023-07-11:
Could you provide a dump of these parameters, ideally matching the values from the web interface? I haven't upgraded mine yet, not sure if should do so. There seems to be no changelog after all.

michischatz, 2023-07-30:
Hi, I had a bit issues to generate a dump, but now its working, but no value is matching to "Eingesetze Energie". This values are calculated by other values? I will try to compare my current dump with the parameters in this library. Will get back to this asap.

kbabioch, 2023-09-28:
I had a somewhat detailed look at those newly added parameters (see #138). It's not obvious to me how these are to be interpreted yet, so further research / observation will be needed.

@michischatz Did you do any follow up investigation here?

michischatz, 2023-09-29:
@kbabioch No sorry, I havent done further investigation here because of summer time.
It seems that the values are calculated because I havent found any 1:1 match from the display to the dump.
Maybe we should consider to rework the process of loading the parameters, calculations and visibilies to the library. A default format like csv or xml would be good to compare the dumps and also version handling of the controller version could be included. As we notice, there are differences between the versions of the controller.

kbabioch, 2023-09-29:
> Maybe we should consider to rework the process of loading the parameters, calculations and visibilies to the library.

If you have specific ideas on how to improve it, feel free to raise another issue. We're happy about good input ;-).

> A default format like csv or xml would be good to compare the dumps and also version handling of the controller version could be included.

It would definitely help. On the other hand I can also compare the current output with difftools, such as `meld`. But of course, improvements are always welcome :-).

> As we notice, there are differences between the versions of the controller.

That is fatal. So far I was assuming that with newer versions only new parameters are added. If parameters change their meaning in between versions, that we would indeed need some kind of version mapping ...

BenPru, 2023-09-29:
> Did you do any follow up investigation here?

See params
[1136 -> HEAT_ENERGY_INPUT -> "Eingesetzte Energie Heizung"](https://github.com/BenPru/luxtronik/blob/4cb8ff1ce5caae441d67e196c475a078df8aedc6/custom_components/luxtronik/sensor_entities_predefined.py#L480)
[1137 -> DHW_ENERGY_INPUT -> "Eingesetzte Energie Warmwasser"](https://github.com/BenPru/luxtronik/blob/4cb8ff1ce5caae441d67e196c475a078df8aedc6/custom_components/luxtronik/sensor_entities_predefined.py#L539C23-L539C39)

gerw, 2023-10-06:
> > As we notice, there are differences between the versions of the controller.
> 
> That is fatal. So far I was assuming that with newer versions only new parameters are added. If parameters change their meaning in between versions, that we would indeed need some kind of version mapping ...

In my opinion, there are no indications that some parameters have changed their meaning.

romixch, 2024-04-09:
It would be so great to see the parameters 1136 and 1137 on my home assistant. I see it is already committed. So I will wait for the release to get them. In the mean time I maybe create a template sensor to get the value

BenPru, 2024-04-09:
> parameters 1136 and 1137 on my home assistant

@romixch 
If you would like to have it in HA, you can use also my integration: [BenPru/luxtronik](https://github.com/BenPru/luxtronik)
See sensor.luxtronik*_heat_energy_input and sensor.luxtronik*_dhw_energy_input
Prehaps this is a faster solution for you.

michischatz, 2024-12-22:
> See params [1136 -> HEAT_ENERGY_INPUT -> "Eingesetzte Energie [...]

To understand you: That means it is already accessable with through the parameters?
https://github.com/Bouni/python-luxtronik/blob/main/luxtronik/parameters.py
Line 1183 und 1184:
            1136: Unknown("Unknown_Parameter_1136"),
            1137: Unknown("Unknown_Parameter_1137"),

Maybe it make sense to rename it?

Guzz-T, 2025-12-23:
Has this issue been resolved with #204?