Bouni/python-luxtronik #112: Calculation 259 and Cooling

gerw, 2023-07-11:
I would like to guess that calculation 259 is related to cooling:

* 0 means cooling is not active
* 1 means cooling is active

Today, I turned on cooling and I was surprised that calculation 80 stayed at "no request" and did not turn to "cooling". However, calculation 259 switched from 0 to 1.

Btw: for me, calculations 117-119 always give a 0 for me. (Even if I go to the heat pump and select the screen where these three lines pop up at the display.)

gerw, 2023-07-11:
This is the setting for calculation 259:
````
DAT_003d6a0c = (uint)Signal_CoolingClearance;
````
Hence, it might not indicate that the cooling is running, but at least that the cooling is allowed?

kbabioch, 2023-07-11:
> Btw: for me, calculations 117-119 always give a 0 for me. (Even if I go to the heat pump and select the screen where these three lines pop up at the display.)

At least for me those three calculations are currently:

> Number: 117   Name: ID_WEB_HauptMenuStatus_Zeile1                                Type: MainMenuStatusLine1  Value: heatpump running
Number: 118   Name: ID_WEB_HauptMenuStatus_Zeile2                                Type: MainMenuStatusLine2  Value: since
Number: 119   Name: ID_WEB_HauptMenuStatus_Zeile3                                Type: MainMenuStatusLine3  Value: heating

I'm running `V3.85.8`.

kbabioch, 2023-07-11:
> Hence, it might not indicate that the cooling is running, but at least that the cooling is allowed?

Seems so. Do you happen to have the decompiled code for 258 also? It's also unknown, maybe it's also related (just guessing).

gerw, 2023-07-12:
> At least for me those three calculations are currently:

Yes, but these are exactly the transformed values of the raw value `0`.

> I'm running `V3.85.8`.

I am at `V3.85.6`.

@all: Does anybody get a value of `7` aka `cooling` for the calculation 80 `ID_WEB_WP_BZ_akt`? My heatpump is cooling and this calculation is still at `5` aka `no request`.

kbabioch, 2023-07-12:
For me the values currently are:

```
Number: 80    Name: ID_WEB_WP_BZ_akt                                             Type: OperationMode        Value: cooling
```

So, yeah, it is `cooling`.

However the other values (117-119) are still "wrong":

```
Number: 117   Name: ID_WEB_HauptMenuStatus_Zeile1                                Type: MainMenuStatusLine1  Value: heatpump running
Number: 118   Name: ID_WEB_HauptMenuStatus_Zeile2                                Type: MainMenuStatusLine2  Value: since
Number: 119   Name: ID_WEB_HauptMenuStatus_Zeile3                                Type: MainMenuStatusLine3  Value: heating
```

The display is showing `Kühlbetrieb`.

So probably this is another case for something that needs to be corrected on a "higher level" (similar to #84).

RobinFlikkema, 2023-07-12:
> @ALL: Does anybody get a value of `7` aka `cooling` for the calculation 80 `ID_WEB_WP_BZ_akt`? My heatpump is cooling and this calculation is still at `5` aka `no request`.

Nope, Value 80 is `5`; Eventhough the RBE shows it is actively cooling and the screen of the heatpump shows `Koelbedrijf`.
117-119:
```
Value 117: 0
Value 118: 0
Value 119: 0
```
This is using `V3.85.8`

gerw, 2023-07-12:
@kbabioch @RobinFlikkema : Which kind of heat pump do you have? Does your heat pump uses the compressor for cooling?

I have a brine/water heat pump which does not use the compressor for cooling. Looking at the firmware, it might be the case that calculation 80 can only turn 7 if the compressor is used for cooling...?

kbabioch, 2023-07-13:
> @kbabioch @RobinFlikkema : Which kind of heat pump do you have? Does your heat pump uses the compressor for cooling?

I have a LWAV 82R1/3, which is an air to water heat pump. It uses the compressor to cool down the water running through the heating system.

So your assumption might be correct. Potentially this is something that should be reported to Alpha Innotec. On the other hand the parameters are not an official feature anyway. Maybe we can reproduce this in some other interface (BACNet, Modbus)?

gerw, 2023-07-13:
In the "offical" web server of the heat pump, there is `Betriebszustand` (operation mode) under `Anlagenstatus` (status of the machine). If my heat pump is cooling (without using the actual heat pump), it does not display anything; for you it shows "cooling". So this is not a real bug for AIT, since the actual heat pump is not cooling.

gerw, 2023-07-13:
I do not see any easy way to check whether the (passive) cooling (i.e., without the heat pump) is running. The safest way seems to be to check whether the calculation 66 `ID_WEB_Zaehler_BetrZeitKue` increases over time.

kbabioch, 2023-07-13:
> I do not see any easy way to check whether the (passive) cooling (i.e., without the heat pump) is running. The safest way seems to be to check whether the calculation 66 ID_WEB_Zaehler_BetrZeitKue increases over time.

Which might be a valid approach for the high level interface. A function / property like `get_operation_mode` could return the currently interpreted operation mode. Considering the runtime of calculation 66 could be handled internally by storing the last known value and comparing it, etc.

RobinFlikkema, 2023-07-14:
> @kbabioch @RobinFlikkema : Which kind of heat pump do you have? Does your heat pump uses the compressor for cooling?

Brine/Water. That would explain it. It's an Alpha Innotec 62K3M