Bouni/python-luxtronik #117: Add @gerw as contributer / maintainer of this project

kbabioch, 2023-07-13:
Since we don't have a formal way to communicate with each other, I'm suggesting this openly as an issue:

@gerw has recently contributed a few interesting improvements and seems to be quite familiar with programming (and reverse engineering).

Therefore I'm suggesting to add him as maintainer to this project, so he can also formerly review merge requests and approve them, etc.

Bouni, 2023-07-14:
I totally support your suggestion as my spare time is even more limited at the moment than it was a few months ago.
That will help this project move forward for sure!

Any thoughts on better ways of communicating between us maintainers?

Bouni, 2023-07-14:
I've sent an invite to @gerw

kbabioch, 2023-07-14:
> Any thoughts on better ways of communicating between us maintainers?

A chat would be ideal, but I'm not familiar with current solutions. Back "in my day" we would use IRC for that, but now there seem to be more web-based solutions available.

Bouni, 2023-07-14:
If we go for a chat, I would prefer matrix over a commercial cloud service as Discord or Slack.
I could self host it on one of my servers if theres no instance out there where we can create a channel for us ...

Any thoughts?

kbabioch, 2023-07-14:
Let's continue the discussion about chats in #121. Since @gerw has been invited, let's close this issue?

closed 2023-07-14 by kbabioch