Bouni/python-luxtronik #121: Solution for developers to interact in more real-time

kbabioch, 2023-07-14:
Currently we only have GitHub available and communicate via issues and pull requests. It might be beneficial to discuss some aspects such as features and who is going to work on what more interactively in a chat.

This issue can be used to discuss options and decide on which solution to use.

Bouni, 2023-07-14:
I've setup a Matrix instance (not completely working yet) which looks good to me.
The only "rule" I want to have expressed before we use that is, that we only use the chat for non public discussions that should not take place within the issues. Everything else must be done via issues to keep the development as transparent as possible.

Let me know what you think!

Bouni, 2023-07-14:
@kbabioch @gerw @BenPru 

Feel free to register an account over at https://matrix.bouni.de/

I have the registration open for now but will disable it as soon as all of you have an account.
I think it should be possible to join with an existing matrix account on another instance but I have not tried that.

Bouni, 2023-07-14:
Update: I enabled Login via Github, that should give us Logins like `@kbabioch:matrix.bouni.de` for example

Bouni, 2023-07-14:
@kbabioch I've sent you an invite, you should be able to accept it and then enter the room