Bouni/python-luxtronik #130: Get rid of _read_after_write()

Guzz-T, 2023-07-21:
The connection to the heat pump is currently being established within the `_read_after_write()` function.

This makes it difficult to offer additional functions such as `read_parameters()`.

kbabioch, 2023-07-22:
Currently struggling a little bit to understand the use case here. So the socket should be opened when the class is initiated, then there is a bunch of read and/or write methods, and only when the class is de-initialized the socket should be closed.

So rather than opening the connection in `_read_after_write()` it should be opened upon initialization.

Is this correct?

Guzz-T, 2023-07-23:
The intention behind this change is different from what you think.

Nothing should change in the previous behavior. Instead of burying the code to connect in the `read_after_write()` function, I moved it to a decorator.

The socket is still opened when the object is used for the first time and closed during freeing. It is also ensured that the critical section is only called by a maximum of one thread.

The change means that the code for the functionality described above can not only be used for `read()` and `write()`, but that you can also provide additional functions. There it must also be ensured that only one thread accesses it and that the socket has been opened beforehand.

PR #131 (2024-02-05): Added a decorator to connect to the heat pump
Using a decorator we can simply re-use the code to connect to the heat pump. This allows us to offers several read / write functions like only read the paramters.

Fixes #130
closed 2024-02-05 by Bouni