Bouni/python-luxtronik #132: Parameters (and friends) should be members of Luxtronik

gerw, 2023-07-22:
One of the changes of pull request #86 was to remove `parameters`, `calculations` and `visibilities` as members of `Luxtronik`. This change was not needed to arrive at the goal of #85 to support multiple instances of `Luxtronik` (this was solved by making the parameters instance members of `Parameters`).

In order to ease the implementation of a high-level interface for reading (see #119), I suggest to add back `parameters` as an instance member of `Luxtronik`. Otherwise, I do not see an easy way for providing functions like `get_firmware_version` to `Luxtronik`.

kbabioch, 2023-07-22:
Are only `parameters` needed/wanted as part of `Luxtronik`, or also the other references to `calculations` and `visibilities`?

Also: Should we work on this right away for [milestone](https://github.com/Bouni/python-luxtronik/milestone/1), or only "later" when working on the high-level interface (https://github.com/Bouni/python-luxtronik/milestone/2)?

gerw, 2023-07-22:
Definitely, we also need `calculations`. Concerning `visibilities`, I do not really know what these tell me.

In my opinion, it fits either milestone, but (if we do it), it should be done before the next release (since the current behavior breaks the API and this issue proposes to revert this).

kbabioch, 2023-07-22:
> Definitely, we also need `calculations`. Concerning `visibilities`, I do not really know what these tell me.

I would then argue that for the sake of consistency we should treat them equal, until we have a good reason to treat them differently. So let's keep a reference to all of those "data vectors".

> In my opinion, it fits either milestone, but (if we do it), it should be done before the next release (since the current behavior breaks the API and this issue proposes to revert this).

Currently the milestones are not linked to releases, but that should be the goal. So let's add it to milestone 1, which is concerned about refactoring, etc.

Guzz-T, 2023-07-23:
I'm still of the opinion that an (outdated) image of the data has nothing to do with the interface.

if you want to save the (actually negligible) time for creating the `parameters` objects, we could also offer an `update()` function, in which the read data is written into the passed object.

```
def __init__():
    self.parameters = Parameters()
    self.luxtronik = Luxtronik("192.168.1.1")

def read ():
    self.luxtronik.update(self.parameters)

...
```

gerw, 2023-07-23:
> I'm still of the opinion that an (outdated) image of the data has nothing to do with the interface.

In my opinion, the `Luxtronik` class is more than a read-write interface. Of course, the data might be outdated (if not re-read frequently), but this is the case for (almost) all the data on your computer ;)

> A "higher-level-function" could looks like:
> 
> ```python
> def get_hot_water_temperature(temperature):
>     param_queue = ParamQueue()
>     param_queue.set(ID_HOT_WATER_TEMPERATURE, temperature)
>     data = self.write(param_queue)
>     read_temperature = data.get(ID_HOT_WATER_TEMPERATURE)
>     return read_temperature
> ```

But with such an approach, you have to read the parameters from the heat pump for each call of a "higher-level function"...

> Has anyone ever tried to read a single parameter instead of all together? This could be useful for "higher-level-functions".

This is not possible (i.e., not supported by the heat pump interface). You can only read all parameters at once.

Guzz-T, 2023-07-23:
> But with such an approach, you have to read the parameters from the heat pump for each call of a "higher-level function"...

This would be the same if you refresh the cached data. Just returning the cached data instead of re-reading it again would be the same as:
``` python
params = luxtronik.get_parameters()
luxtronik.high_lvl_func_without_reading()
foo = params.get(ID)
```
No need to cache this inside the interface.

> This is not possible (i.e., not supported by the heat pump interface). You can only read all parameters at once.

Not possible due to the Python interface or the Luxtronik control itself?

gerw, 2023-07-23:
> This would be the same if you refresh the cached data. Just returning the cached data instead of re-reading it again would be the same as:
>
> ```python
> params = luxtronik.get_parameters() [...]

What is achieved by the second line here? How should `high_lvl_func_without_reading` execute, when it does not have any data available.

> > This is not possible (i.e., not supported by the heat pump interface). You can only read all parameters at once.
> 
> Not possible due to the Python interface or the Luxtronik control itself?

This is not possible by the firmware of the heat pump.

Guzz-T, 2023-07-24:
> > ```python
> 
> > params = luxtronik.get_parameters()
> 
> > luxtronik.high_lvl_func_without_reading()
> 
> > foo = params.get(ID)
> 
> > ```
> 
> 
> 
> What is achieved by the second line here? How should `high_lvl_func_without_reading` execute, when it does not have any data available.

This is just to clarify that nothing is read back there. In principle, high-level functions with (`SetHotWaterTemp(50)`) and without parameters (`ActivateHolidayMode()`) would be possible

gerw, 2024-02-09:
Just a quick reminder: If revert the API change, we should also call `read()` in `__init__` again to load the current state of the heat pump.

Guzz-T, 2024-02-10:
I still vote not to add the data to the interface.

We could implement something like this:

```python
LuxtronikInterface:
  read() -> LuxtronikData
  write(...) -> LuxtronikData

LuxtronikData:
  __init__():
    self.parameters = ...
    self.visibilies = ...
    self.calculations = ...
  
  some_high_level_func():
    ...

Luxtronik:
  __init__():
    self.Interface = LuxtronikInterface()
    self.Data = self.Interface.read()

  some_further_high_level_function()
    ...
    
```
(Did not checked the syntax)

Guzz-T, 2024-02-10:
https://github.com/Guzz-T/python-luxtronik/tree/issue/132/revert-api

Here is a suggestion. Due to the base classes there is a little more code, but a cleaner separation of the interface from the data.

Bouni, 2024-02-12:
@Guzz-T I just had a quick look at your "revert-api" proposal and so far this looks good to me.
Not breaking the existing API is a big bonus in my eyes, which your proposal fulfils, right?

I have not yet test your branch due to lack of time and I can't before wednesday.

@gerw and @kbabioch any thoughts on this?

gerw, 2024-02-12:
Do we really need this level of abstraction? Yes, it seems to be nice from a design point of view. But it blows up the code and I do not think that it improves the maintainability. I also think that (besides the socket connection) there will be no other way of obtaining data from the heat pump in the future.

I like the simplicity and the API of `0.3.14`. From my point of view, the only circumstance where the current API is better, is if you want to keep a history of the parameters. With the new API, one can use
````python
l = []
while True:
  p = lux.read_parameters()
  l.push_back(p)
````
whereas the old API needs
````python
l = []
while True:
  lux.read_parameters()
  l.push_back(copy(lux.parameters))
````

Guzz-T, 2024-02-12:
@gerw you are right. It's quite a lot of code for little functionality. I have updated the branch with a new proposal.

That would be a compromise of compatibility, simplicity and a clear separation of the interface from the data.

```python
class LuxtronikData:
...
class LuxtronikInterface:
...
class Luxtronik(LuxtronikData, LuxtronikInterface)
```

gerw, 2024-02-13:
Yes, this looks nice. I have a few comments:

* Should `Luxtronik` really be inherited from `LuxtronikInterface`? From my perspective, it would be more logical, if it is a member, i.e., `self.interface = LuxtronikInterface(...)`, or rather `self._interface`?
* That `Luxtronik` is inherited from `LuxtronikData` seems to be ok, since we can call `luxtronik.some_high_level_function()` without additional gluing code.
* In the old API,
````python3
lux = Luxtronik(ip, port)
p = lux.parameters
...
lux.read()
# Use "p" to access the parameters, or just:
assert p == lux.parameters
````
would work, but it fails with your code. How about something like
````python3
class LuxtronikInterface:
    def read(self, parameters = Parameters()):
        [...]
        return parameters
class Luxtronik:
    def read(self):
        return self.interface.read(self.parameters)
````
?
* Although I suggested otherwise, I am now wondering if `LuxtronikInterfaceBase` would be a good idea. We could implement some dummy interface(s) which would just present some data to add more test routines (and get more code coverage). In `LuxtronikInterfaceBase` we could add the methods `_with_lock_and_connect`, `read`, `read_*`, `write`, `_read`, `_read_*`, `_write`, since these are pretty generic.  Child classes have to implement `_connect`, `_disconnect`, `_send_ints`, `_read_int` and `_read_char`.

* Typing seems to be nice, but it should not be introduced while working on these changes. This should go to a different PR.

Guzz-T, 2024-02-13:
Opened pull-request #163

@gerw:
- `LuxtronikInterface` is now a member of `Luxtronik`
- What would be the use case for comparing whether the two address pointers point to the same address (`p == lux.parameters`)? To compare the data `__eq__()` would be required. Sure, it's not one and the same instance, but that shouldn't have any negative effects. If you really have to, we can add this “update” functionallity.
- For now, `LuxtronikInterfaceBase` is not needed for this issue. But could be added with another pull-request.
- Removed typing

gerw, 2024-02-13:
Concerning `p == lux.parameters`: One use case is that we save the reference to `lux.parameters` at some place in our code. Then, we could call `lux.read()` somewhere else (e.g. in a separate thread) every minute to update the parameters, which can still be accessed by `p`.

[In my opinion, it is also not necessary to create a new `Parameter` object for each reading, but the old one can be recycled.]

Guzz-T, 2024-02-13:
Now you answered too quickly. I would have revised the comment again.

Guzz-T, 2024-02-13:
Please have a look again. Alternatively `DataVector` itself could provide a `update(self, other_data_vector)`.

gerw, 2024-02-14:
Let's continue the discussion in the PR.

Guzz-T, 2024-02-20:
Please note that even after #163, this code is not a 1-to-1 replacement for the previous version 0.3.14.
I still get a lot of errors when starting homeassistant.

Examples:
```txt
2024-02-20 21:38:38.291 WARNING (MainThread) [Luxtronik.Calculations] entry 'ID_WEB_SoftStand' not found
2024-02-20 21:38:38.304 WARNING (MainThread) [Luxtronik.Calculations] entry 'Unknown_Calculation_239' not found
AttributeError: 'PoolMode' object has no attribute 'measurement_type'
    File "/config/custom_components/luxtronik/coordinator.py", line 322, in firmware_version_minor
        return int(ver.split(".")[1])
IndexError: list index out of range
```

gerw, 2024-02-20:
* `ID_WEB_SoftStand` is no longer there, should be replaced by a high-level-function; this also causes the IndexError
* Calculation 239 now has a real name `VBO_Temp_Spread_Soll`
* `measurement_type` is now `datatype_unit`?

Guzz-T, 2024-02-20:
Changes seem logical. However, they must be implemented by the user (BenPru/luxtronik). This means that no direct exchange is possible.

PR #163 (2024-02-22): Issue-132: Revert API changes introduces with #86
With #86 we split the data from the read/write interface.
However, these changes mean that the existing program needs to be adjusted.

With this pull-request we restore the previous api by:
- Add a data class that contains the parameters, calculations and visibilities
- Keep the seperation of the data class and the read/write interface
- Provide a new "Luxtronik" class that combines the data and the interface

This allows both the old and the new API to be used.

Fix #132 

NOTE: The linter added some format changes.
closed 2024-02-22 by Bouni