Bouni/python-luxtronik #135: Introduce and maintain a changelog

kbabioch, 2023-07-22:
We should introduce and maintain a changelog that is meant to be read by users of this library. It should focus on notable changes (new features, etc.).

A good example is "Keep a Changelog", which is described in more detail [here](https://keepachangelog.com/en/1.1.0/).

I don't think it is worthwhile to create it retroactively for previous releases, but with a upcoming new (major) release we should start to use it and keep the notable changes prepared in an `Unreleased` section and have a good description as part of (bigger) pull requests.

Bouni, 2023-07-24:
I totally agree with this!
I've read somewhere that there's kind of a workflow to make it easier by prefix PRs with [feature], [bug] and so on which then can be parsed by an GH action to lower the hurdle keeping the Changelog up to date.
Unfortunately I don't recall where I've seen that 😅

Bouni, 2023-07-26:
@kbabioch I searched but didn't find the article I've read in the past. But I stumbled up on [Git Cliff](https://git-cliff.org/)
I guess something like that is the way to go. For sure we then have to stick with [conventional commits](https://www.conventionalcommits.org/en/v1.0.0/) but that can also be enforced with a GH action via [commitlint](https://commitlint.js.org/#/) as well as a pre-commit-hook I guess

Guzz-T, 2026-03-15:
#240 is the first step towards addressing this issue.