Bouni/python-luxtronik #136: New pypi release

BenPru, 2023-09-12:
Hello heatpump lovers,
heating season is near. Although my time is short, I would like to advance the new code base of the HA component.
If someone would like to support, you can install the [preview](https://github.com/BenPru/luxtronik/releases/tag/2023.9.11-Alpha-Core-Preview) or [make code review for the core module](https://github.com/home-assistant/core/pull/86636).

For the new code I use [luxtronik==0.3.14](https://github.com/BenPru/luxtronik/blob/b0a58a558e9e125c061169458019715ca79450e4/custom_components/luxtronik/manifest.json#L13) and the new functions with an [own Luxtronik class](https://github.com/BenPru/luxtronik/blob/2ef45b2e95f84786424a0ea772378db955ebdf3a/custom_components/luxtronik/lux_helper.py#L161). (But time.sleep is not allowed in a core module...)

It would be great if we could get a new stable version here that I can then use.
What is missing for a new stable version?

FnTm, 2023-10-04:
Sorry about the additional ping @Bouni or @kbabioch, but I'm also really looking forward to a new pypi release. Can we help in some way to get that going?

gerw, 2023-10-04:
Yes, this would be nice. But I think that there are some issues which should be addressed before a new release, e.g., #132 and maybe also #135?

kbabioch, 2023-10-04:
I will take another look at #135.

FnTm, 2024-01-11:
@Bouni or @kbabioch sorry to bother you folks, but is there some way to get the new pypi release made?

Bouni, 2024-01-15:
Hey everyone, sorry for the long silence from my side but personal things had (still have) priority over hobby projects 😓
I actually lost tack of what we added after 0.3.14 and whats necessary to release a stable 0.3.15 or even a 0.4.0

@kbabioch @gerw @BenPru what do you guys think is necessry for a new release. Can we might just release a intermediate 0.3.15 version now and a lets say 0.4.0 when the features are complete we inteded for the next "big" release?

Bouni, 2024-02-09:
@gerw Do you think #132 is a must have for a 0.3.15 release?

If not I would like to release 0.3.15 and we can plan on having #132 for the 0.3.16 or whatever release in the future.

gerw, 2024-02-09:
@Bouni At least we should discuss, how we handle that issue. Currently, `main` has a breaking API change compared to 0.3.14 and that issue proposes to revert this breaking change. From my point of view, it does not make sense to release 0.3.15 with this braking change and then revert it in 0.3.16.

[The API change is that `parameters` is no longer a member of `Luxtronik`. Instead `read()` returns the `parameters`.]

gerw, 2024-02-19:
@Bouni Now, #163 is almost finished. After merging, I propose to deal with #128 (should be a quick one). It would also be nice, to start a basic changelog, see #135.

Actually, what has been changed since 0.3.14?
* Identified datatype of many parameters / calculations
* Internal refactoring
* The connection to the heat pump is reused (instead of connect & disconnect for each read/write operation)
* TBD: New way to access the software version (#128)

Bouni, 2024-02-19:
I guess we have to do the changelog for a new version manually this time. Afterwards I tend to introduce [commitlint](https://commitlint.js.org/#/) / [conventionalcommits](https://www.conventionalcommits.org/en/v1.0.0/) to ensure commit messages that allow for a semi-automatic changelog generation.

What do you guys think about this?

gerw, 2024-02-20:
Guzz-T found some incompatibilities to 0.3.14 in #132. These should also mentioned in the change log.

I have no experience with this, but we could try conventionalcommits. Do I get it right, that every commit will end up in the change log?

gerw, 2024-02-21:
I took the liberty to create a new milestone for the next release. This way we can focus on the necessary changes to publish the release in the next time.

gerw, 2024-04-01:
ping @Bouni @kbabioch

rhammen, 2025-08-30:
@Bouni @kbabioch @gerw 
Any updates on when a new release (0.3.15) will be available?
The HomeAssistant integration  BenPru/luxtronik  relies on this library, and things are getting more and more difficult when e.g. new heatpump codes  and new parameters are appearing that are not supported by this library....
It would be great if you could just release a new version with these kind of things updated. :-)
Please?

Bouni, 2025-09-01:
@gerw @kbabioch I know we had great plans to get this super clean but I feel we are blocking to much in the mean time.
Maybe we should release a intermediate version and save big stuff for a future (0.4!?) release.

I will try to test the current state of the main branch against my HA setup. If that works I would like to release a 0.3.15

Any thoughts?

@BenPru I would really like not to break your HA Plugin with a new release. Do you have time to test the main branch as well?

rhammen, 2025-09-04:
@Bouni Thanks for your response, happy to hear that it may be possible to get a 0.3.15 version out in the not too distant future!

The BenPru/luxtronik integration currently uses "requirements": ["luxtronik==0.3.14",], so a new version should not immediately break things. 
I will try to setup a branch in the BenPru integration where we use the current Bouni/python-luxtronik@main branch, so we can learn if there are any issues.