Bouni/python-luxtronik #138: Power Consumption (available in 3.88.0)

woozar, 2023-09-28:
The service tech updated my luxtronik to 3.88.0 and since that, I get these items in the web ui:
 
[image]

Is it possible to read them through this plugin? What are the ids here? 

If not, how can I assist, so we can add those values together?

kbabioch, 2023-09-28:
Providing a dump using the dump functionality might be helpful to see what the (raw) data looks like and whether new datatypes will be needed. I also want to update my heatpump, but wasn't sure if there are any negative side effects. Having full access (root, etc.) and this library working is nice, so I don't want to loose that ;-).

kbabioch, 2023-09-28:
By the way: This is essentially a duplicate of #100.

woozar, 2023-09-29:
@kbabioch that is true - i did go through that list before creating my ticket, but I didn't realize it. 
Ass the dumps are posted here, I will not close this issue, but I will also not be mad, if somebody else does.

BenPru, 2023-09-29:
See params
[1136 -> HEAT_ENERGY_INPUT -> "Eingesetzte Energie Heizung"](https://github.com/BenPru/luxtronik/blob/4cb8ff1ce5caae441d67e196c475a078df8aedc6/custom_components/luxtronik/sensor_entities_predefined.py#L480)
[1137 -> DHW_ENERGY_INPUT -> "Eingesetzte Energie Warmwasser"](https://github.com/BenPru/luxtronik/blob/4cb8ff1ce5caae441d67e196c475a078df8aedc6/custom_components/luxtronik/sensor_entities_predefined.py#L539C23-L539C39)

kbabioch, 2023-10-02:
> See params [1136 -> HEAT_ENERGY_INPUT -> "Eingesetzte Energie [...]

When comparing this to the values on the heatpump itself, it's probably even more specific, i.e. it's the "Eingesetzte Energie" from the heat pump itself (vs. from the heating rod):

[image]
[image]
[image]

Probably the parameters next to it are related to the other energy input data (cooling, etc.), but this will need more investigations to figure it out.

gerw, 2023-10-06:
I think that parameter 1138 is related to the swimming pool and 1139 is for cooling.

Does anybody know how precise these values are (e.g. if your heat pump is also connected to a power meter)?

stoxn, 2023-10-15:
Hi gerw, I just compared the values / deviation after seven days:
- Smart meter consumption: 32 kWh
- Consumption as per Novelan: 28.8 kWh

Smart meter is read via an infraread interface; it measures power consumed by Novelan + circulation.

As my dedicated smart meter for the heating system will be removed while installing PV, I would be happy to read the consumption values. My target is to read and analyse values to get notifications in case of deviations from normal behaviour and to calculate the JAZ (even though it will be a little better).

kbabioch, 2023-10-15:
> As my dedicated smart meter for the heating system will be removed while installing PV, I would be happy to read the consumption values. My target is to read and analyse values to get notifications [...]

You can still get another meter installed (after the official one), in order to get more reliable data. Seems like there is a ~10% deviation, which makes the values for JAZ not reliable. As you've said there is also other consumers to consider (e.g. circulation pump, etc.).

gerw, 2023-10-16:
Thank you. During the installation of PV, my official smart meter for the heat pump was removed, but my electrician installed me a 'dumb' meter (with S0 interface) for the heat pump.

When my heat pump is idling, the meter is showing a power consumption of 15-16W. However, also the 11 room sensors (they are no longer connected to the heating system...) are attached to this meter and I suspect that a significant amount of power is drawn by these room sensors. What is your meter showing for an idling heat pump?

stoxn, 2023-10-16:
Thanks for your feedback, both!

Indeed I asked my Electrician to install a separate meter with S0 and/or Modbus, so I can read reliable consumption data.
My current smart meter is read via impulses, minimum unit 1 kWh. So I never measured idle consumption...

Room temperature sensors are read via KNX; the room actuators are powerless fully opened. But at least the circulation will consume a decent amount of energy.

I'll keep an eye on the heater values for comparison.

BenPru, 2023-10-16:
>  What is your meter showing for an idling heat pump?

~11 Watt.

BenPru, 2023-10-16:
> I think that parameter 1138 is related to the swimming pool and 1139 is for cooling.

Is this knowledge certain?
Do you know what 1140 can be?

gerw, 2023-10-17:
> ~11 Watt.

Is this only/exactly the heat pump or is something else connected to the meter?

gerw, 2023-10-17:
> > I think that parameter 1138 is related to the swimming pool and 1139 is for cooling.
> 
> Is this knowledge certain? Do you know what 1140 can be?
Yes, I am rather certain. It would be nice if it could be confirmed by somebody who has reasonable entries in these parameters for his heat pump.

1140 could be the the amount of energy used for the second heat source when preparing hot water (as shown in the pictures by kbabioch.

BenPru, 2023-10-17:
> Is this only/exactly the heat pump or is something else connected to the meter?

Only HP I think. If the compressor heater or a pump (solar, dhw/heat circulation,...) is working the consume rises.

RobinFlikkema, 2023-11-25:
Did you guys confirm this?

As I was looking and for my Luxtronik controller, these positions don't seem to line up.
This is what the webinterface shows:
[image]
However, I am only finding parameter _1059: 4912_
I wasnt able to find the 0.7kWh nor the 491.9kWh in any of the output

BenPru, 2023-11-25:
Yes, it is 1137. The newer socket params "energy inputs" have a delay. It seems that the websocket gets the values faster.