Bouni/python-luxtronik #143: Calculation ID 267: desired room temperature

RobinFlikkema, 2023-11-25:
I would like to open this issue, as I currently seem to have found Calculation 267. It would be great if others could confirm this.

`267 : 2147483647`

[image]

This is in Dutch: Gewenste Ruimtetemperatuur, translating to Desired Room Temperature.

The value itself is a bit weird (showing the max int), but maybe others have this value set to something else and can confirm these findings.

RobinFlikkema, 2023-12-02:
> I don't have calc 267. My max is calc 262. I have a Novelan LAD7 with newest Firmware [V2.88.3-9717](https://www.heatpump24.com/software/fetchSoftware.php?softwareID=1)

Ah, i didn't specify the version but mine is running 3.89.3

Bouni, 2024-01-15:
@RobinFlikkema 

I can confirm that I see the same value with V3.89.4

```
Number: 267   Name: Unknown_Calculation_267             Type: Unknown              Value: 2147483647
```

RobinFlikkema, 2024-01-18:
> @RobinFlikkema
> 
> I can confirm that I see the same value with V3.89.4
> 
> ```
> Number: 267   Name: Unknown_Calculation_267             Type: Unknown              Value: 2147483647
> ```
> 
> But the value seems strange to me!? Is that if no value is set or somethig like that?

Yeah, however this is also showing in the webinterface. Furthermore, this is the max 32-bit integer. Maybe this only works if you have a specific setup? with Alpha Home for example?

Nonetheless, we at least know that these two values (the one in the calculation_267 and in the webinterface) are the same.

Bouni, 2024-01-19:
@RobinFlikkema Still we don't kno wwhat the value means. To me this seem like a bug that this value shows up in the webinterface.
I would expect a normal value for desired room temperature like 20.5°C and not a bazillion °C 😂
Maybe it disappears with the next release when they realize that it only should sho up when it has a plausible value or so ...

closed 2024-02-05 by gerw