Bouni/python-luxtronik #165: socket.recv waits infinite when the connection is lost

Guzz-T, 2024-02-14:
If the connection is already established, a new call to `read()` will never finished in case the connection to the heat pump is lost inbetween.

It seems to resume if the heat pump is available again.

Is this a bug or a feature?

gerw, 2024-02-15:
We could set [`_socket.setdefaulttimeout()`](https://docs.python.org/3/library/socket.html#socket.setdefaulttimeout) to a sensible value, e.g., 5 seconds.

What shall we do when the timeout occurs? Raise an error? Try to reconnect?

Guzz-T, 2024-02-15:
Normally I would close the socket and return the error status using the return parameter. The user then has to handle the error.

But that's almost like an "API change".