Bouni/python-luxtronik #173: PiP installtion shows strange behaviors

strike19, 2024-04-01:
I am not that firm with git, but I followed the guidance from your readme and installed the package via:

pip install luxtronik

The installation seems successful with the following reply:
>Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
>Collecting luxtronik
>Installing collected packages: luxtronik
>Successfully installed luxtronik-0.3.14

When I am trying the code examples from your page the following error occurs:
>File "lux_solar.py", line 4, in <module>
>    from luxtronik import Luxtronik
>  File "/home/pi/.local/lib/python2.7/site-packages/luxtronik/__init__.py", line 8, in <module>
>    from luxtronik.calculations import Calculations
>  File "/home/pi/.local/lib/python2.7/site-packages/luxtronik/calculations.py", line 313
>    calculation = Unknown(f"Unknown_Calculation_{index}")

I have checked manually the code and it seems that the version installed via pip differs from the version on this githubpage and all installed modules have the same typo calculation = Unknown(**f**"Unknown_Calculation_{index}")

Did I miss something when I installed the packacke or is this due to the fact that I am working on a Raspberry Pi?

Any help would be appreciated.

PS: On the long run, I would activate the function "1052" - "ID_Einst_Warmwasser_extra", but it seems that this is parameter has no options...

gerw, 2024-04-01:
It seems that the actual error is missing. Can you show the full error message?

I would guess that your python version does not understand the f-string? Which version are you using?

Do you know what parameter 1052 is doing? Which values are possible?

gerw, 2024-04-02:
Yes, this package needs python3.

I do not think that this menu is related to parameter 1052. If I am not mistaken, parameter 1052 can be accessed via `Service -> Einstellungen -> Inverter -> Warmwasser' and it can take the values "Normal" (0) or "Luxus" (1). With "Luxus", the heat pump uses a higher frequency to heat the water.

Do you know in which version precisely these parameters where introduced?

In the `main` branch of the repository, you can find some convenience scripts (https://github.com/Bouni/python-luxtronik/tree/main/luxtronik/scripts). Can you try to run `dump_luxtronik.py`? If you use the old pip version, you might have replace the constant `LUXTRONIK_DEFAULT_PORT` by `8889`, otherwise the script should also run with the older version. Then, it would be really helpful if you can run the script, afterwards make some settings in this menu and run the script again. Can you post the output (and the settings you have made)? [Note that the output also contains the `ID_WebserverPasswort`, which, of course, could be obfuscated.]

strike19, 2024-04-02:
My installed firmware is: V3.89.3

I have used the dump_changes script and could identify the following parameters:
para: Number: 1147  Name: Unknown_Parameter_1147                                       Value: 5400 -> reverted
para: Number: 1146  Name: Unknown_Parameter_1146                                       Value: 650 -> 600

==> 1147 "Dauer Extra WW" in seconds and 1146 "Sollwert" in °C

The rest does not make any sense to me... find below the dump with "on" and "off" with my guesses in "**bold**"

I hope that helps :-)

calc: Number: 249   Name: Unknown_Calculation_249                                      Value: 319 -> 361
calc: Number: 178   Name: ID_WEB_LIN_UH                                                Value: 44.900000000000006 -> 7.5
calc: Number: 250   Name: Unknown_Calculation_250                                      Value: 355 -> 396
**calc: Number: 38    Name: ID_WEB_BUPout                                                Value: False -> True**
calc: Number: 80    Name: ID_WEB_WP_BZ_akt                                             Value: no request -> hot water
calc: Number: 43    Name: ID_WEB_VBOout                                                Value: False -> True
calc: Number: 184   Name: ID_WEB_HZIO_VEN                                              Value: 0 -> 415
calc: Number: 57    Name: ID_WEB_Zaehler_BetrZeitImpVD1                                Value: 20369 -> 20370
calc: Number: 236   Name: ID_WEB_Freq_VD_Soll                                          Value: 0 -> 3000
calc: Number: 237   Name: ID_WEB_Freq_VD_Min                                           Value: 0 -> 30
calc: Number: 238   Name: ID_WEB_Freq_VD_Max                                           Value: 0 -> 100
calc: Number: 44    Name: ID_WEB_VD1out                                                Value: False -> True
calc: Number: 56    Name: ID_WEB_Zaehler_BetrZeitVD1                                   Value: 50314845 -> 50314942
calc: Number: 63    Name: ID_WEB_Zaehler_BetrZeitWP                                    Value: 50314845 -> 50314942
calc: Number: 65    Name: ID_WEB_Zaehler_BetrZeitBW                                    Value: 17595798 -> 17595895
calc: Number: 67    Name: ID_WEB_Time_WPein_akt                                        Value: 1 -> 98
calc: Number: 72    Name: ID_WEB_Time_SSPEIN_akt                                       Value: 0 -> 1103
calc: Number: 231   Name: ID_WEB_Freq_VD                                               Value: 0 -> 2970
calc: Number: 257   Name: Heat_Output                                                  Value: 0 -> 4949
calc: Number: 179   Name: ID_WEB_LIN_UH_Soll                                           Value: 20.0 -> 7.9
calc: Number: 252   Name: Unknown_Calculation_252                                      Value: 900 -> 1150
calc: Number: 152   Name: ID_WEB_WMZ_Brauchwasser                                      Value: 23498.9 -> 23499.100000000002

calc: Number: 249   Name: Unknown_Calculation_249                                      Value: 319 -> 359
calc: Number: 178   Name: ID_WEB_LIN_UH                                                Value: 44.900000000000006 -> 21.6
calc: Number: 250   Name: Unknown_Calculation_250                                      Value: 355 -> 425
**calc: Number: 38    Name: ID_WEB_BUPout                                                Value: False -> reverted**
calc: Number: 80    Name: ID_WEB_WP_BZ_akt                                             Value: no request -> reverted
calc: Number: 43    Name: ID_WEB_VBOout                                                Value: False -> reverted
calc: Number: 184   Name: ID_WEB_HZIO_VEN                                              Value: 0 -> reverted
calc: Number: 57    Name: ID_WEB_Zaehler_BetrZeitImpVD1                                Value: 20369 -> 20370
calc: Number: 236   Name: ID_WEB_Freq_VD_Soll                                          Value: 0 -> reverted
calc: Number: 237   Name: ID_WEB_Freq_VD_Min                                           Value: 0 -> reverted
calc: Number: 238   Name: ID_WEB_Freq_VD_Max                                           Value: 0 -> reverted
calc: Number: 44    Name: ID_WEB_VD1out                                                Value: False -> reverted
calc: Number: 56    Name: ID_WEB_Zaehler_BetrZeitVD1                                   Value: 50314845 -> 50315031
calc: Number: 63    Name: ID_WEB_Zaehler_BetrZeitWP                                    Value: 50314845 -> 50315031
calc: Number: 65    Name: ID_WEB_Zaehler_BetrZeitBW                                    Value: 17595798 -> 17595977
calc: Number: 67    Name: ID_WEB_Time_WPein_akt                                        Value: 1 -> reverted
calc: Number: 72    Name: ID_WEB_Time_SSPEIN_akt                                       Value: 0 -> 942
calc: Number: 231   Name: ID_WEB_Freq_VD                                               Value: 0 -> reverted
calc: Number: 257   Name: Heat_Output                                                  Value: 0 -> reverted
calc: Number: 179   Name: ID_WEB_LIN_UH_Soll                                           Value: 20.0 -> reverted
calc: Number: 252   Name: Unknown_Calculation_252                                      Value: 900 -> reverted
calc: Number: 152   Name: ID_WEB_WMZ_Brauchwasser                                      Value: 23498.9 -> 23499.100000000002
calc: Number: 141   Name: ID_WEB_Time_AbtIn                                            Value: 4580 -> 4444
[... cut ...]
calc: Number: 113   Name: ID_WEB_Switchoff_file_Time2                                  Value: 2024-04-02 17:47:47 -> 2024-04-02 19:33:26
calc: Number: 114   Name: ID_WEB_Switchoff_file_Time3                                  Value: 2024-04-02 19:33:26 -> 2024-04-02 21:18:02
calc: Number: 115   Name: ID_WEB_Switchoff_file_Time4                                  Value: 2024-04-02 21:18:02 -> 2024-04-02 21:38:52
calc: Number: 71    Name: ID_WEB_Time_SSPAUS_akt                                       Value: 0 -> 229
calc: Number: 29    Name: ID_WEB_ASDin                                                 Value: True -> reverted