Bouni/python-luxtronik #176: Parameter details 1034/1035/1041/1043/1045/1052

gndlff, 2024-04-19:
For extending the Hotwater and Heating Control via Homeassistant, i've made some investigation regarding the following parameters and their settings and valid ranges:

Model: AIT SWCV62k3
Firmware: 3.85.8

1034: ID_Einst_P155_PumpDHWCtrl
Luxtronik Menu Entry = Effizienzpumpe > Steuerung WW. 
Sets the pump control mode during hot water generation
1 = Automatic (Pump is regulated to achieve a fixed temperature spread between supply and return)
0 = Manual (Pump is set to fixed speed parameter 1035)

1035: ID_Einst_P155_PumpDHW_RPM
Luxtronik Menu Entry = Effizienzpumpe > Steuerung WW. 
Sets the fixed speed of HUP during hot water generation
value 1 - 100 equals setting 1 - 100% in Luxtronik

1041: ID_Einst_P155_PumpDHW_Max
Luxtronik Menu Entry = Effizienzpumpe > Max. Lstg. WW 
value 1 - 100 equals setting 1 - 100% in Luxtronik
The valid range that can be set in the Luxtronik is 50 - 100%. Via this interface, lower values can also be written. 
I use this  to reduce disturbance in the Hot Water Boiler without throttling the flow with valves.

1043: ID_Einst_P155_Sperrband_1
Defines the invalid operating frequency range of the inverter 
Luxtronik Menu Entries: Inverter > Sperrband Start / Inverter > Sperrband Stop
Both settings in Luxtronik modify the same value. The values are mapped as a sum of both parameters:

Sperrband Start: value 65536 - 7864320 equals setting 1 - 120 Hz
Sperrband Stop: value 1 - 120 equals setting 1 - 120 Hz

Both value are added for the final parameter value.

Example: 
Sperrband Stop 30 Hz = 1966080
Sperrband Start 23 Hz = 23
ID_Einst_P155_Sperrband_1 = 1966080 + 23 = 1966103

1045: ID_Einst_P155_DHW_Freq
Luxtronik Menu Entry = Inverter > Freq. WW 

Allows a fixed speed for Hot Water Generation (can be used to mimic the "Target Temperature" Hot water generation mode of the Nibe Heatpump Controllers in combination with parameter 1035)

This parameter is usually only writable with service level access on the luxtronik, but can be written via the interface without issue.

value 0 is Automatic
value 1 - 101 equals setting 20 - 120 Hz

1052: ID_Einst_Warmwasser_extra
Luxtronik Menu Entry = Inverter > Warmwasser
value 0 = Normal
value 1 = Luxus (increased compressor frequency during hot water generation)

kbabioch, 2024-04-19:
Interesting findings, thank you for your effort.

Could you explain how exactly you're making use of those parameters for instance? Does it make sense to change those values from their defaults?

gndlff, 2024-04-19:
It really depends on the hot water system i think.

I use a fresh water station for hot water genration which is fed by a 500L tank of heating water. There is no heat exchanger between Heatpump and Hot water storage. By default, the Luxtronik is running the HUP pump at maximum speed during the first few minutes of a hot water genration cycle and the end of the cycle. This causes a lot of disturbance in the different hot water layers in the tank, causing the hottest layer at the top getting mixed with colder water and thus reducing the output temperature. 
By limiting the parameter 1041 to a value of 20-30%, this effect is reduced. Depending on the strategy even values as low as 5% make sens for greater spread between supply and return and higher positioned feed in connections to the storage tank.
This behaviour can also be reached by throttling the flow via a valve and setting it to the lowest possible setting in luxtronik, but then the pump in running at higher speed to overcome the increased resistance consuming more energy.

A further problem in the default control mode with fixed spread of 7K is that when a large storage tank is used, there are long cycles for fully loading the tank with a small heatpump (4kW heating power during hot water genration). I use two sensor for controlling the hot water genration. One at the top of the tank and one at the heatpump feed in in the lower third of the tank. Hot water cycle is started when the upper sensor is below 43°C and runs until the lowe sensor has reached 40°C. Theoretically the supply temperature of the heat pump never needs to go over +-43°C to achieve that. Due to the fixed spread of 7K of the luxtronik, the feed in temperature however increases up to 50°C when the tank is almost fully loaded, decreasing efficiency and increasing wear of the compressor.

By changing parameter 1045 (decreasing it from the usual frequency after the target temperature 45°C has been reached) the maximum temperature is limited. This logic is done in home assistant.

Advantage of using parameter 1045 for this instead of frequency limit 1043 is that you can still use frequency limit for limiting heating power in heating mode and have fewer parameter changes in the controller memory.

Hot gas temperature needs to be monitored to avoid triggering overtemperature faults when playing with the settings for compressor and pump speed.

tepee1, 2024-04-19:
Hi gndlff,

thanks for your input, veeery interesting. We have recently met in another forum :)

I also have a Luxtronik controlled heatpump, but with 180l integrated hot water tank. This has some other challenges too, e.g. the tank is emptied quite rapidly. especially when using bath tub. 

So far I have manually tuned my hot water process facing same topics as you have described above too:

-At the beginning HW pump at 1-2% + allowing higher inverter speed up to 70-80Hz in order to receive enough power, thus high temp. spread and soon hot water at the top. Normally I have limited my inverter at 55Hz during winter time or 40Hz during summer. In the winter 55Hz can be too low for rapid hot water processing, at least at the beginning when HW tank temp is quite low. In the summer playing just with pump speed could be sufficient.

-Later with increasing temp in the tank the challenge is to avoid high feed in temp. With my small HW tank I use up to 48 deg HW target temp. This results with the stock 7K spread in receiving 55 deg feed in temps. With manual slow pump speed even higher. So the pump speed has to be increased accordingly (or also inverter speed reduced during winter), in my case to stay below 50 deg.

Somehow to mimic this - similar to Nibe - target temp HW charging using Home Assistant would be great. I can't understand Luxtronik has this not implemented. At Nibe this came also later via an update some years ago. I have asked technical people at the manufacturer, in my case Roth, but they don't seem to care about this. 

Luxtronik is missing some other functions too, e.g. direct button for more hot water etc.

Also the additional temp sensor for top tank temperature is very useful but not available at Luxtronik.
Where did you put this and how did you implement this additional temp sensor at top of the tank?

gerw, 2024-04-24:
I agree with your observations.

> **1043: ID_Einst_P155_Sperrband_1** Defines the invalid operating frequency range of the inverter Luxtronik Menu Entries: Inverter > Sperrband Start / Inverter > Sperrband Stop Both settings in [...]

Your first description is the wrong way round. The higher bits describe "Sperrband Stop". The example at the end is correct.

> **1052: ID_Einst_Warmwasser_extra** Luxtronik Menu Entry = Inverter > Warmwasser
> value 0 = Normal
> value 1 = Luxus (increased compressor frequency during hot water generation)

In fact, "Luxus" increases the frequency by "20" (seems to be hardcoded).

kbabioch, 2026-01-01:
From my point of view this is something that could be worked on. Probably some new datatypes might be needed.

@gndlff any new insights since when you originally reported this?