Bouni/python-luxtronik #178: SmartGrid

riess82, 2024-05-21:
Dear all.

Great module and great work on collecting all those values and settings.

I am trying to connect heatpump and solarpanels in a "smart" way.
Using a microcomputer, some python code and relais, I can set the EVU1 and EVU2 so that the heatpump knows what to do.

However, I was unable to find the value displayed as "Smart Grid" in the heatpump or the EVU2 in python-luxtronik so far. I want to read the setting back from the heatpump to confirm my relais did a good job. A dump of the changes has not displayed anything useful, I can see ID_WEB_EVUin change, which is great, but I can't find the second value.

I think my goal is quite similar to https://github.com/Bouni/python-luxtronik/issues/173 but I will use EVU1 and EVU2 in SmartGrid mode to turn the heatpump higher or lower.

Any help in finding either the SmartGrid value (0 to 4) or EVU2 would be great.

gerw, 2024-07-12:
EVU1 seems to be in calculation 31 and EVU2 could be in calculation 185.

In which menu can you see "Smart Grid" or "EVU2"?

riess82, 2024-07-12:
In the webserver "Smart Grid" is under "Informationen", "Eingänge". EVU1 and EVU2 are not visible in the webserver as separate values. Only combined into "Smart Grid". I'll test with the parameters and calcs mentioned above.

gerw, 2024-07-12:
What do you mean with "combined"? Can you show a screenshot of this menu of the web server?

dasSofa, 2024-07-12:
He probably means this.
[image]

riess82, 2024-07-12:
exactly. here is what the value means:
# 1 off
# 2 low
# 3 normal
# 4 high