Bouni/python-luxtronik #189: Make lookup of names easier for users

Bouni, 2024-11-18:
I think its not very easy for the user to figure out what name / id to use in the HA config.

@Guzz-T made that point in the discussion in #187 , that we should write the code in a way that makes it easy for the user to look the name up. 

I propose that we add a GitHub Action that parses the code instead and create a static web page with a search field that we then host on GitHub pages for the users to look the right parameters up.
That is in my eyes the better way to help the user than to write the code in a way that users can look up the needed nmes directly in the code.

gerw, 2024-11-18:
I think we should also add some small piece of documentation for each parameter / calculation for which we know the meaning.

Bouni, 2024-12-11:
@gerw I agree, my key point here is that the user should not have to look up the needed info in the source.

regetskcob, 2025-05-08:
Hi there, I'm just starting to setup my HomeAssistant to communicate with our Novelan Heatpump and I am also struggling with the parameter-ids and their meaning. Wouldn't be a quick-win to simply add kind of a translation-/mapping-table to the repo's ReadMe where the id is mapped to the English and/or German meaning? 

For example I am currently searching for my solar-heating fore- and and back-run temperatures. It's very painful to understand which of https://github.com/Bouni/python-luxtronik/blob/main/luxtronik/calculations.py could mean what I am looking for.