Bouni/python-luxtronik #190: Support SmartHomeInterface?

gerw, 2024-11-18:
Since Version V3.90.1 of the luxtronik controller, there exists a so-called SmartHomeInterface. It can be enabled in the settings of the controller (or via parameter 1161, I think). It listens via Modbus-TCP on Port 502 and 'controls' the registers 10.000-19.999 (only a few of them are in use currently). There are some registers for reading (input registers) and others for reading/writing (holding registers). Via the holding registers, it is, e.g., possible to control the "desired return flow temperature". The advantage is that the written values are not saved in the persistent memory of the controller, i.e., these writing operations should not add wear to the NAND memory, see #158.

Should we add access to this interface to this library?

gerw, 2024-11-18:
No, of course there is no documentation. I also asked the support of alpha innotec, but they refused to provide some documentation.

The interface is used by the [SolarManager](https://www.solarmanager.ch).

The new interface uses the same port as the "old" ModBus interface, but it uses other registers.

Guzz-T, 2025-03-07:
> Can you do something via the ModBus interface which is not possible via port 8889 (this library)?

Currently, the heating circuit/water heating / mixing circuit temperatures and the power limitation can be set via the SHI.
This was previously only possible by rewriting the configuration in the NAND memory (e.g., fixed value temperature).

I have started implementing something for this. I believe this additional interface would be an enrichment for this library:
https://github.com/Guzz-T/python-luxtronik/tree/try/smart-home-interface

kbabioch, 2025-03-07:
Sounds / looks very interesting. Do you have any documentation / mapping of the registers involved?

kbabioch, 2025-03-07:
Seems like some information can be found [here](https://www.haustechnikdialog.de/Forum/t/284442/Eigene-Regelung-PV-Luxtronik-2-1-Smart-Home-Interface-SHI), at least for German reading users.

Guzz-T, 2025-03-07:
Unfortunately not. Documentation is scarce at AIT. I extracted most of the information from the linked thread.

Guzz-T, 2025-03-10:
Here you will find an updated version of the Home-Assistant integration:
https://github.com/Guzz-T/bouni-luxtronik/tree/try/smart-home-interface

Together with the latest state of the branch from above, the smart-home-interface values should be readable by HA.

felixs2103, 2025-06-26:
> Here you will find an updated version of the Home-Assistant integration: https://github.com/Guzz-T/bouni-luxtronik/tree/try/smart-home-interface
> 
> Together with the latest state of the branch from above, the smart-home-interface values should be readable by HA.

I can provide the Modbus Datapoint Documentation i got from Novellan (luxtronic) if it helps.

kbabioch, 2025-09-13:
> Today I integrated the latest data fields from version 3.92.0. Now there are some interesting functions:
> 
>     * Start hot water preparation immediately
> 
>     * Lock heating / hot water
> 
>     * Control of the circulation pump
> 
>     * and a lot of unkown stuff
> 
> 
> Are you interested in integrating this into this repository?

Yes, definitely interested! Could you hint us to the place where you've already integrated this? Obviously pull requests are also highly welcome, but a good reference implementation is also useful.

Guzz-T, 2025-09-13:
Here are the links (or see above):  
Reference implementation:  https://github.com/Guzz-T/python-luxtronik/tree/try/smart-home-interface
HA component update: https://github.com/Guzz-T/bouni-luxtronik/tree/try/smart-home-interface

I could try merging the changes in small pieces via pull requests.