Bouni/python-luxtronik #195: Write values to ID_Einst_Effizienzpumpe

TheSteiny, 2025-04-11:
Hi

is it possible to write Pararmeter ID_Einst_Effizienzpumpe_akt or ID_Einst_Effizienzpumpe_Nominal_akt ?
I tried both with Option safe=False, but got the following error:
Parameter id 'xxx' or value 'x' invalid!

Thx!

gerw, 2025-04-11:
What is the precise error message?

TheSteiny, 2025-04-11:
parameters.set("ID_Einst_Effizienzpumpe_akt", 0) returns:
Parameter id '869' or value '0' invalid!

gerw, 2025-04-11:
This cannot be true. `parameters.set(...)` cannot give this error message:
```python
    def set(self, target, value):
        """Set parameter to new value."""
        index, parameter = self._lookup(target, with_index=True)
        if index is not None:
            if parameter.writeable or not self.safe:
                self.queue[index] = parameter.to_heatpump(value)
            else:
                self.logger.warning("Parameter '%s' not safe for writing!", parameter.name)
        else:
            self.logger.warning("Parameter '%s' not found", target)
```

TheSteiny, 2025-04-13:
Got it to work. There was a issue in my script. value was paresed as string instead of number. 
So i can confirm writing values to ID_Einst_Effizienzpumpe_akt or ID_Einst_Effizienzpumpe_Nominal_akt works smoothly.
luxtronik 0.3.14

closed 2025-04-14 by gerw